olikraus/u8g2 #892: otf2bdf issue when converting ttf to bdf

waylander131, 2019-05-12:
trying to convert ttf file to bdf then to c++ file.

when attempting the first conversion got error "otf2bdf: unable to open temporary file '/tmp/otf2bdf13544'."

found old issue 2017 from user Nakonzaid who got the same problem and later reported that he got it sorted but not sharing how.

Any help will be appreciated.

[image]

olikraus, 2019-05-12:
I am not the author of otf2bdf and I run otf2bdf under ubuntu, which runs without problems.
Idea. Create a "c:\tmp". Maybe this helps.

closed 2019-05-19 by olikraus

Project-Magenta, 2021-11-06:
For reference of future users, you just have to make a folder called "tmp" in the root of whatever drive you are running from

DARIOF8, 2023-07-03:
@Project-Magenta I have the same problem I don't know how to tell otf2bdf where my font is. can you explain in detail how it works and how to use it, please? 
[image]

krishna116, 2026-08-12:
当使用otf2bdf工具时， 我遇到了类似的问题：unable to open temporary file '/tmp/otf2bdf13544'。
我检查了源代码，问题是作者默认使用 "/tmp/otf2bdf"+pid 当作临时文件名字，这种临时文件路径在windows系统中是无效的。
我的解决的方式是修改源代码(otf2bdf.c)，将临时文件名字改成 "tmp-otf2bdf"+pid 格式。
以下是改良后的代码：
    if ((tmpdir = getenv("TMPDIR")) == 0)
      tmpdir = "tmp";
    sprintf(tmpfile, "%s-otf2bdf%ld", tmpdir, (long) getpid());
    if ((tmp = fopen(tmpfile, "w")) == 0) {
        fprintf(stderr, "%s: unable to open temporary file '%s'.\n", prog, tmpfile);
        return -1;
    }