olikraus/u8g2 #893: LM6063 not ported to u8g2lib

alanmi3833, 2019-05-14:
Using LM6059 constructor with u8glib, I get a mirrored screen, with text barely visible.
Using LM6059 constructor with u8g2lib, I get just a white screen, can't see any text.

All other ST7565 constructors, gives only blue screen.

Regards

olikraus, 2019-05-14:
Which interface? Argument/pin sequence might be different.

alanmi3833, 2019-05-14:
I am not sure I understand, but using software SPI.
U8GLIB_LM6063 u8g(sck, mosi, cs, a0 , resetlcd);

alanmi3833, 2019-05-14:
This is the constructor I use with u8g2lib:
U8G2_ST7565_LM6059_F_4W_SW_SPI u8g(U8G2_R0, sck, mosi, cs, a0, resetlcd);

olikraus, 2019-05-14:
Looks good. Can you run the u8g2 contrast example? Will you see something?

alanmi3833, 2019-05-14:
I can definitely see text being displayed from 0 to about 50 contrast. Although the frames looks more like a T being displayed. Text are also in the lower left corner, so looks like it count lines bottom to top.

So I will have to see why it doesn't doesn't display when using my program where I just change the constructor from U8G to U8G2

olikraus, 2019-05-14:
I think the LM6063 just requires a different contrast setting compared to other displays.

My suggestion is this: Use the setContrast command and apply a value between 10 to 30 and retest the existing constructors with a standard HelloWorld example.

alanmi3833, 2019-05-15:
Will need to do more testing, seems to be working with LM6059 if I set the contrast. Will reopen if I can't get it to work
Thank you for the help.

olikraus, 2019-05-15:
ok

alanmi3833, 2019-05-22:
LM6059 doesn't work, prints gibberish on left hand side of screen.
However the following construct does work:
`U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI u8g(U8G2_R0, sck, mosi, cs, a0 , resetlcd);`
with a contrast of:
` u8g2.setContrast(50) //Value between 40 and 60 works great.` 

Reason I couldn't get it to work with u8g2 were because I use the SD library.
It seems that a lot of people have a problem with SD and u8g2. SD works perfectly with u8g. Any pointers where to start looking why u8g2 does not work with SD, while u8g works.

Regards

olikraus, 2019-05-22:
Use HW SPI, see FAQ ;)

olikraus, 2019-05-22:
Gibberish: 

You should test each of the constructors. Remember to adjust also the contrast for each of the displays. In fact the contrast value might be different. At least something must be visible with any of the constructors.

U8G2_ST7565_EA_DOGM128
U8G2_ST7565_64128N
U8G2_ST7565_ZOLEN_128X64
U8G2_ST7565_LM6059
U8G2_ST7565_LX12864
U8G2_ST7565_ERC12864
U8G2_ST7565_ERC12864_ALT
U8G2_ST7565_NHD_C12864
U8G2_ST7565_JLX12864

Which of the constructors will show a picture without gibberish?
Is there none? If you found at least one constructor I will create a new constructor with the correct contrast value for your display.

alanmi3833, 2019-05-22:
The constructor above together with setcontrast does work without a problem.
I have put maybe to much info in one comment.
Here are the constructor again which works 100%.
`U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI u8g2(U8G2_R0, sck, mosi, cs, a0 , resetlcd);
u8g2.setContrast(50) //Value between 40 and 60 works great.`

olikraus, 2019-05-22:
ah, ok, I missunderstand your post. 
I will create a new constructor, based on the DOGM128 with a default value 50 for the contrast.

olikraus, 2019-05-22:
Will keep this open until I created the constructor

olikraus, 2019-05-22:
New constructor prefix: `U8G2_ST7565_LM6063`

olikraus, 2019-05-22:
I have created a new beta 2.26.10.
Maybe you can test the new constructor...
Note, that the constructor is not part of the examples, it is listed here only: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lm6063

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

alanmi3833, 2019-05-23:
Olikraus, the LM6063 constructor are there, but it seems that the contrast were not updated.
Doing the following to get it to work
`U8G2_ST7565_LM6063_1_4W_SW_SPI u8g(U8G2_R0, sck, mosi, cs, a0 , resetlcd); 
  u8g.setContrast(50); //Have to set this as well
`

olikraus, 2019-05-23:
ah... yes, thanks. Created beta 2.26.11

alanmi3833, 2019-05-24:
Thanks it is working now.

olikraus, 2019-05-24:
Great. Thanks for testing :-)

closed 2019-06-12 by olikraus