olikraus/u8g2 #894: Help identifying Pins 320x240 LCD

almamater, 2019-05-14:
Hello. Can you please confirm if my old LCM320240 - SED1335 works with this library? I cant understand and match the pins of the LCD to the library\example. My lcd pinout: https://i.postimg.cc/s2dzKj53/PinOut.png 

Datasheet: http://qingyun-qingyunit.b176.vhostgo.com/products/yejing/simply/tx/LCM3202401.pdf

Is there any chance of using this display?

I thought on this combination but i think is not correct..:

**My LCD**=========> **Library\Arduino**
CS=Chip Select(6) =====>          RS (10)
A0 Data (7 )=========>           RW(11)
RD Read Data (5)======>         E (13)
Reset (8) ===========>      Reset (8)

Thank you!!

olikraus, 2019-05-15:
It should work, connection depends on the interface selected for your display.
If it is a 8080 interface, then the connection should be:
your display ---> u8g2
CS=Chip Select(6) =====> cs
A0 Data (7 )=========> dc
RD Read Data (5)======> Vdd
WR ======> enable
Reset (8) ===========> Reset (8)

I also assume, that you need to feed back Vout into V0 via an variable resistor.

almamater, 2019-05-15:
Ok, many thanks for your reply! I'll try wiring it it that way.

closed 2019-05-19 by olikraus