olikraus/u8g2 #895: Crash on Particle Boron

picsil, 2019-05-17:
I know this has been addressed with no resolution previously in a now closed issue, but I'm also trying to get the Arduino version of u8g2 working on a Particle Boron. Upon u8g2.begin() the MCU crashes every time. I'm using hardware SPI.

I've stepped through the debugger, and the crash always happens at u8g2->cb->update_dimension() in u8g2_SetBufferCurrTileRow(). I have not been able to trace into update_dimension() to further locate the crash.

I am willing to donate a couple of Particle Argon (wifi) or Boron (LTE) MCUs if you'd be willing to develop a fix. I like the versatility of this library, but after spending a few days trying to debug the problem I am stumped.

olikraus, 2019-05-18:
Definitly I am willing to develop a fix, but what I can not do is the debugging part. It will be too time consuming to understand and install this specific development environment to carry out debugging on my own laptops.

If you came so far, can you tell me details about the crash? What kind of crash is it? Memory fault? division by zero? What is the exact line of code or assembler statement?

olikraus, 2019-06-02:
No further input... closing...

closed 2019-06-02 by olikraus