olikraus/u8g2 #897: Drawing creates dark patches on bigger shapes

liamlacey, 2019-05-21:
I've got an issue where drawing shapes / lines / text on one side of the screen will create dark patches on bigger shapes on the other side of the screen. Here are some photo examples of what I mean:

[image]

[image]

[image]

I've tried three different SSD1309 and one SSD1306 OLEDs all using hardware SPI in full screen buffer mode, both on an Arduino uno using the C++ code and using the C code in Atmel Studio, and I get the same results.

The code I'm using to do this is very simple:

```
u8g2.clearBuffer();
 u8g2.drawBox (0, 0, 60, 16);
 u8g2.drawLine (0, 32, 60, 32);
 u8g2.drawBox (64, 0, 64, 64);
 u8g2.sendBuffer();

```
Is this to be expected? 
If not, am I doing something wrong or could this be an issue with the library?

Thanks in advance.

olikraus, 2019-05-21:
It is not related to the library. It is a question of current sink and the internal resistance of your power supply: One light bulb at one power supply is bright, multiple light bulb on the same power supply might be little bit darker. You can see this clearly in your first picture. Just relate light bulbs with individual pixels.

The only thing you can do is to reduce the internal resistence of the power supply. 
You could try to add extra or larger or better caps to the external power supply wires of your display.
If this doesn't help you may need to consider a full external power supply instead of the default charge pump of the display (but this would require more work)

liamlacey, 2019-05-21:
Many thanks for the information - I'll take a look into this.

AnHardt, 2019-05-21:
Usually an other way to reduce the problem is to reduce the needed current to draw a complete line, by reducing the current for every single pixel by reducing contrast/brightness.

closed 2019-06-02 by olikraus