olikraus/u8g2 #898: Issues with i2c_t3 and U8g2 conflict

AndyBasa, 2019-05-21:
I found your great libraries u8g2 after just purchasing a CrystalFontz CFA012864D3-TFH-CB.  I successfully can display graphics and texts - and also manipulating the back light brightness with PWM.  I then need to integrate the display with my other devices (GPS, external SD breakout, BLE device using I2C) and ended up with a conflict with i2c_t3.  I am using Teensy 3.6 - and confirmed all code compiles (migrating from another display) without the:

- #include <U8g2lib.h>" and the 
- U8G2_ST7565_LM6059_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 32, /* data=*/ 28, /* cs=*/ 31, /* dc=*/ 8, /* reset=*/ 7);		// Adafruit ST7565 GLCD 

commands

As soon as I don't comment out those lines - I get a host of errors (see some below) pointing to Wire..  I suspect there is some conflict by the i2C_t3 and the U8g2lib.   

Error listing:
"
C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `i2c0_isr':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.cpp:805: multiple definition of `i2c0_isr'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.cpp:1389: first defined here

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `Print::availableForWrite()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: multiple definition of `i2c1_isr'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:915: first defined here

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `Print::availableForWrite()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: multiple definition of `i2c2_isr'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:915: first defined here

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `Print::availableForWrite()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: multiple definition of `Wire2'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:915: first defined here

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `Wire2' changed from 20 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o to 108 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `Print::availableForWrite()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: multiple definition of `Wire1'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:915: first defined here

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `Wire1' changed from 20 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o to 108 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o: In function `Print::availableForWrite()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: multiple definition of `Wire'

C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:915: first defined here

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `Wire' changed from 20 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\i2c_t3\i2c_t3.cpp.o to 108 in C:\Users\JP\AppData\Local\Temp\arduino_build_703512\libraries\Wire\WireKinetis.cpp.o

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Error compiling for board Teensy 3.6.  "

Here is the code for the i2C

#include <i2c_t3.h>
#if defined(__MK64FX512__) || defined(__MK66FX1M0__)  // 3.5/3.6
#define WIRE1_PINS   I2C_PINS_37_38
#endif

Can someone please assist?

Thanks, 
Andy

olikraus, 2019-05-21:
Independent from the solution, I suggest to use HW_ I2C

olikraus, 2019-05-21:
You can also disable I2C in u8x8.h. maybe this helps.

AndyBasa, 2019-05-21:
I appreciate the suggestions - can you tell me how I would disable the I2C in u8x8.h please?

olikraus, 2019-05-21:
Oh, excause, it is actually U8x8lib.h:

Uncomment this line:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L60

AndyBasa, 2019-05-21:
The location you mentioned is already not commented, so I assumed you wanted me to comment it out.  I did and it generated the same errors.  Can you please clarify if I did what you intended?  Here is what I did..

/* Assumption: All Arduino Boards have "SPI.h" */
#define U8X8_HAVE_HW_SPI

/* Assumption: All Arduino Boards have "Wire.h" */
// #define U8X8_HAVE_HW_I2C   //AB:Commented this out

/* Undefine U8X8_HAVE_HW_SPI for those Boards without SPI.h */

#ifdef ARDUINO_AVR_DIGISPARK

olikraus, 2019-05-21:
Ah yes, Idea is to comment this line.
If this doesn't help, then I don't know what todo

AndyBasa, 2019-05-21:
I am sorry but it was my mistake!  I left a copy of and edited file 'U8X8lib.cpp' in the same directory (src) and it generated the compilation errors.  I then removed that duplicate file and it compiled cleanly.  My code is now working.  I really appreciate the quick assistance and have a great day.

Andy

olikraus, 2019-05-21:
:)

closed 2019-05-21 by AndyBasa