olikraus/u8g2 #899: How do display a number in u8g2?

huuchinh02, 2019-05-23:
Hi everyone,

I am using u8g2 library on my project. I have a problem that I can't find any functions to display a number in C.
I have to use u8g2_uint_t u8g2_DrawStr(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, const char *s);
It is very inconvenience for me to convert from number to string and from string to number every time I want to update a new value.
How does the method faster?

Thanks!

olikraus, 2019-05-23:
You can use functions like itoa or sprintf which should be part of your environment already. 
If you really do not have any other number to string conversion functions, there are two functions available in u8g2:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L956

huuchinh02, 2019-05-25:
> You can use functions like itoa or sprintf which should be part of your environment already.
> If you really do not have any other number to string conversion functions, there are two functions available in u8g2:
> 
> https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L956

Yes, many thank olikraus. I must convert number to string using sprintf. By the way, I want to ask you one question.
I use your library to create my digital clock. I used the method below:
`u8g2_FirstPage(pDisp);
do
{
  //I create my GUI here
  // Display second
  //Display min, hour...
} while (u8g2_NextPage(pDisp));`

You know, the second will be updated after 1 second period and I try to check something have a change and just control them but I can't do that because of your lib always put all buffer to oled. I think that not necessary to do that because oled driver allows you to change any specified pixel. This made the blocking code time increase that I did not expect.
Details:
1. When I use 128 bytes of RAM the blocking code time is 20ms
2. When I use 1024 bytes of RAM the one is approximately 9ms

And I measure the time of your lib to send 1024 byte to the driver is 5ms when using SPI 4.5Mhz frequency.
It really consumes most my resource of MCU, it makes another task was be delayed
How can I improve that? I want the loop of the main process below 1ms
My MCU is stm32f103 and I am running at 72Mhz (maximum of clock).

Thanks!

olikraus, 2019-05-25:
Because you use plain C code, I assume you have your own low level driver. Performance depends a lot on the low level driver, so you should ensure, that you make use of your SPI (I2C?) hardware interface.

Another option is to send only a part of the the screen content to the display. I mean, you can only update the upper part for example.
There is an Arduino example here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/UpdatePartly/UpdatePartly.ino

So you probably need to translate this to plain C. The trick is to use [image].

It is a little bit tricky, bit I hope you get the point from the manual and the examples. If not... ask here...

huuchinh02, 2019-05-25:
I ensure to use SPI, because I disable SPI config in low level it doesn’t run. I using your callback send data to my spi send function. You tell true that the performance depends on low level data handle but it’s not too much, I’m using 2 spi for the project and the another spi works well.
As mentioned in previous comment, I measured the time which send buffer to low level (u8g2_nextPages), it spend 5ms too much for my application.
About setBufferCurrTileRow function
I am looking forward it. I has just seen Direct Access Buffer section in your reference
I will try something and ask you if I can’t understand any point.
Thank you!

olikraus, 2019-06-20:
Closing for now. Feel free to reopen.

closed 2019-06-20 by olikraus