olikraus/u8g2 #902: Only first line printing

bill-orange, 2019-05-28:
I can get my OLED display to work but only the first line will display (I think this is about between 8 to 12 vertical pixels).

I am using a generic noname OLED.  This one:

[https://www.ebay.com/itm/US-0-96-I2C-IIC-Serial-128X64-LED-OLED-LCD-Display-Module-for-Arduino-White/382557870211?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649](url)

I am using the latest release of the library.  It is really hard to figure out which descriptor to use!

This one appears to be correct:

`U8G2_SSD1306_128X64_NONAME_1_SW_I2C display(U8G2_R0, SCL, SDA, U8X8_PIN_NONE); `

My function looks line this:

  ```
 display.clearBuffer();
  
    display.drawStr(0, 0, tempF_st);
    display.drawStr(0, 12, humidity_st);
    display.drawStr(0, 24, eco2_st);
    display.drawStr(0, 36, etvoc_st);
    display.drawStr(0, 48, base_st);

   display.sendBuffer();
```

The display is not defective. It works fine with the similar (but not as nice) Adafruit library.

I can get the Heltec OLEDs to work fine, but not this one.

Am I missing something?  Wrong descriptor?  Thoughts?

olikraus, 2019-05-28:
Use the `_F_` constructor instead of the `_1_` constructor.

bill-orange, 2019-05-28:
Yes, that works!  I am a bit curious. What is actually happening here?  What is the difference between the two constructors?

olikraus, 2019-05-28:
https://github.com/olikraus/u8g2/wiki/setup_tutorial#u8g2-full-buffer--page-buffer-and-u8x8-mode

closed 2019-05-28 by bill-orange

olikraus, 2019-05-28:
no problem. Let me know if you have any further problems.