olikraus/u8g2 #903: u8g2 not work with TinyGPS++

fatfenders, 2019-05-29:
I have found that if I comment out the u8g2 code in this sketch (particularly the "send buffer" command), I get valid heading data from Tinygps++, but if I enable the "send buffer" the heading always shows a zero heading.
_______________________________________________________________________________________________
`// Date Written: May 5,2019
// Author: Dave Garner
// Base code is from  Yvan @ https://Brainy-Bits.com
// Processor: Teensy LC
// Input Adafruit Ultimate GPS
// Output: 1.5" SPI OLED display

// This code resides in my '36 Chevy Street rod and is used to display the following
// information on a 1.5" OLED display mounted above the interior rear view mirror.

// Direction of travel
// Elevation
// Direction and distance to my home.

#include <U8g2lib.h>  // U8g2 Library for Oled https://github.com/olikraus/u8g2
#include <Math.h>
#include <TinyGPS++.h>
TinyGPSPlus gps;

char CSats[3] ="00";
char CAlt[5];
char CDis[5];
char CAngle[4] ="xxx";
long IAngle ;
long IAlt;
const char *cardinalHeading ;
int StringPixilLength ;
char HomeString[10];  

// static const double Home_LAT = 38.896565, Home_LON = -121.07689;   // test (Aubrun CA)

  static const double Home_LAT = 38.652807, Home_LON = -121.153539;   // my home location

// U8g2 Library SPI, Software Serial, Full Frame Buffer mode 

//                                              rotation, clock, data, cs, dc [, reset]) 
   U8G2_SSD1327_MIDAS_128X128_F_4W_SW_SPI  u8g2(U8G2_R0 ,   5,    6,   4,  3  ,   2) ;
    
 

void setup() 
 {
  Serial.begin(9600);
  Serial.println("---- Fire this Baby up! 1------");
  Serial1.begin(9600); 
  
  u8g2.begin();  
  u8g2.setContrast(200);  // 0 to 255
//  delay (5000) ;
  
}
void loop()
 {
//---------------------------------------- Grab and format the satellite info -----------------------------  
//  Serial.print("Serial1.available() ");  Serial.println(Serial1.available());
  while (Serial1.available())
     gps.encode(Serial1.read());
//---------------------------------------- Wait till there are at least 4 sats -----------------------------
/*
  if (gps.satellites.value() < 4)
   {
    sprintf(CSats, "%02lu",gps.satellites.value());
//    Serial.print("satellites.value--");  Serial.println(CSats);  
    u8g2.clearBuffer(); 
    u8g2.setFont(u8g2_font_fub14_tr);
    u8g2.drawStr(00, 30, CSats);
    u8g2.setFont(u8g2_font_fub25_tr); 
    u8g2.drawStr(10, 100, "No Fix!");   
    u8g2.sendBuffer(); 
   }
  else
//---------------------------------------- Got 4+?, grab 'em  -----------------------------  
   {
*/ 
   if (gps.satellites.value() > 4 )
     {  
    sprintf(CSats, "%02lu",gps.satellites.value());
   Serial.print("satellites.value--");  Serial.println(CSats);  
//---------------------------------------- Grab current heading  -----------------------------  
    
     IAngle = gps.course.deg();
      itoa(IAngle,CAngle,10) ;
     Serial.println(CAngle);
    const char *cardinalHeading =  TinyGPSPlus::cardinal(gps.course.deg());
 
    Serial.println(cardinalHeading);
//---------------------------------------- Grab current elevation  -----------------------------       
    IAlt = gps.altitude.feet();    

    itoa(IAlt,CAlt,10) ;
    
//---------------------------------------- Grab distance, course to "home"  -----------------------------  
    unsigned long distanceKmToHome =
    (unsigned long)TinyGPSPlus::distanceBetween(
     gps.location.lat(),
     gps.location.lng(),
     Home_LAT, 
     Home_LON) / 1000;
     distanceKmToHome = distanceKmToHome * 0.621371;     
    itoa(distanceKmToHome,CDis,10) ;
    
     double courseToHome =
      TinyGPSPlus::courseTo(
      gps.location.lat(),
      gps.location.lng(),
      Home_LAT, 
      Home_LON); 

//---------------------------------------- Format the OLED Display -----------------------------
  
    u8g2.clearBuffer();   
         
//----------------------------------------- number of satellites ----------------------------     
     u8g2.setFont(u8g2_font_fub14_tr);    
     u8g2.drawStr(00, 35, CSats);     
     u8g2.drawStr(99, 35, CAngle);
      u8g2.sendBuffer();    
   }
   
 
  }
 
 `

olikraus, 2019-05-29:
Do display and gps share any Wires? Is there a RAM overflow? Which Board do you use?

fatfenders, 2019-05-29:
Teensy LC
Sketch uses 40676 bytes (64%) of program storage space. Maximum is 63488 bytes.
Global variables use 5208 bytes (63%) of dynamic memory, leaving 2984 bytes for local variables.
I have all 3 components (Teensy; OLED; GPS) all together in a small space. No wires are over 4" long.
The 4 signal wires for the OLED are about 1/2" inch. Other than power/ground, the only wires connected to the Teensy are those 4 OLED wires (pins 3456) and pin0 'RX1'.   
And thank you for the super fast reply (was not expecting that!)

fatfenders, 2019-05-29:
[image]
[image]

olikraus, 2019-05-29:
Hmm... ok, so the usual problem of shared wires does not apply. There also seems to be sufficient RAM.
OLEDs do emit some high frequency signals. For testing you could disconnect the OLED but still execute the sendBuffer command. If the GPSs works, then there might be a hardware problem between OLED and GPS.

fatfenders, 2019-05-29:
I removed ground from the OLED and made 2 tests, uploading each time.
test 1. GPS always reported a heading of 113.
test 2. GPS always reported a heading of 356.

Previously the heading would jump around since I am working 'on the bench'.

I had this same problem before I went to the Teensy LC. Was using a Mega.
I got everything working by using a Nano for the Tinygps, then sending it to the Mega over serial. Kinda messy but that actually worked for me. Was hoping for a tidier solution.

fatfenders, 2019-05-29:
Past my bedtime. I will check this thread in the morning. Thanks again.

olikraus, 2019-05-29:
sure... I just started my day....

fatfenders, 2019-05-30:
Update!
I gave up on Tinygps++ and starting over with the Adafruit GPS example 'parse' sketch. I ported over most of my u8g2 code. Bingo! While I have only tested 'on the bench' I seem to consistently be getting good GPS results. 
The Adafruit library appears to be not nearly as robust as TinyGPS but it works with u8g2! The Adafruit library apparently has no "cardinal heading" option so I had to code my own (no big deal). But I do want to get "distance/heading between two points". That is all I have left to figure out since it looks like Adafruit does not have that option either.
So, at least in my case, there is definitely a conflict between Tinygps and u8g2.

olikraus, 2019-05-30:
Hmmm thanks for the feedback.

fatfenders, 2019-06-09:
Just wanted to let you know that I ended up using the NEOgps libraries and it works with well u8g2 plus it supports 'distance between 2 points'.

olikraus, 2019-06-09:
:)

closed 2019-06-20 by olikraus

himikeb1, 2023-09-20:
I was having the same problem. I went through the TinyGPS++ source code and am pretty sure it's NOT TinyGPS - the ONLY code I could find is string and value processing, nothing with I/O.
I believe the problem is the PORT that TinyGPS reads - I think U8g2 conflicts with the I/O used by Tiny.
What I was able to get working was NeoGPS, which has a dependency on AltSoftSerial.  I think AltSoftSerial probably does not conflict with U8g2, so if you could get TinyGPS to read using AltSoftSerial, you'll probably be fine.

olikraus, 2023-09-21:
hmm very old topic. 

> I think U8g2 conflicts with the I/O

U8g2 also does not use any I/O directly. It will use SPI.h, Wire.h and digitalWrite only. 

Conflicts might be there if you connect a display and another device on the same bus like I2C or SPI.
Then a conflict might be there, because U8g2 apply the max speed supported by the display. Usually libs don't decrease the speed for their device, so after high communication with the display, talking to a different device may not work any more.