olikraus/u8g2 #904: u8g2 Font in C#

Bungalore, 2019-05-30:
Greetings everyone
I am working on a simple 128x64 screen layout editor in C# (new for me)

It is meant to show exactly what should be on the real LCD, so using and displaying fonts is crucial.
While searching on the forum, I found this post
https://github.com/olikraus/u8g2/issues/824
And the OP shared fully functionnal python scripts, I want to port them on C#, but unfortunately, I am struggling with proper reading of font format.
The python translation of the original u8g2 lib is for me as clear as the original source code.

Can someone provide me explanation on how it works with actual example, [saying this font for instance](https://github.com/olikraus/u8g2/blob/master/tools/font/build/single_font_files/u8g2_font_4x6_tn.c)

Why there are numbers in font data, mixed with actual characters, with all those backslashes, I am stopping at reading font name :)

My programming skills are average (coming from java and pascal)

Any help is welcome

olikraus, 2019-05-30:
> Why there are numbers in font data, mixed with actual characters, with all those backslashes, I am stopping at reading font name :)

This is just binary data represented as a c string. None ASCII chars are represented as octal escape codes: https://en.wikipedia.org/wiki/Escape_sequences_in_C

You can get a rough idea of the font format from here: https://github.com/olikraus/u8g2/wiki/u8g2fontformat

But finally, you have to understand the innermost for loop, which draws the glyph: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L564

olikraus, 2019-05-30:
BTW: There is also a SDL port of u8g2, so it will run under linux without problem. Maybe you can call the u8g2 C/C++ functions from C#

I mean, this would allow you to use all the u8g2 functions directly in your editor.

Bungalore, 2019-05-30:
Thank you for the reply
The RLE byte interpretation part seems no issue for me, I have a problem at the beginning with converting the c++ binary string into array of bytes in C#, to get data from font.c file.
As C# doesn't interpret escaping characters the same way as C++, I need some decoder to do this, I will look this out and come back once I figure it out. Maybe this is out of the scope of u8g2 library, but anything of help is welcome.
BTW, I am on Windows and Visual Studio 2015, which uses almost unicode only.

olikraus, 2019-05-30:
Hmmm... indeed, it looks like that C# does not support octal escape sequence: https://stackoverflow.com/questions/40201971/showing-octal-characters-in-c-sharp

Bungalore, 2019-06-01:
I have successfully made a C++ escaping interpretor like for C# and finally got bytes of font data.
Performance is not considered ATM, but the algorithm should work pretty fast.

I share it here so anyone using C# can take befit from it.

        public byte[] readCPPDataString(string inStr)
        {
            // splitting input string with fragments of data separated by a backslash, don't question variable name choice :)
            string[] slashes = inStr.Split(new[] { '\\' }, StringSplitOptions.RemoveEmptyEntries);

            // Dynamic list to store actual bytes of data, after interpretation
            List<byte> bytes = new List<byte>();
            
            // Reads fragments of data, one at a time
            for (int i = 0; i < slashes.Length; i++)
            {
                // For more convenience, used a string rather than calling slashes[i] every time
                // For understandability purpose only
                string item = slashes[i];

                // See function delaration for more info
                int lastNumericPos = IsNumeric(item);

                // == -1 means it is is an actual integer value, not checking if there are more than 3 octal digits at the moment
                // This will introduce an error if there is more than 3 digits in input expression (\123[any digit further] gives a value > 255)
                // While it should consider only the 3 first digits as octal and the rest as UTF8-bytes
                // Little chance to happen but has to be fixed
                if (lastNumericPos == -1)
                    bytes.Add((byte)Convert.ToInt32(item, 8)); // Converts to bytes in octal form

                else // The sequence is not / not only digits
                {
                    UTF8Encoding Encoding = new UTF8Encoding();

                    // Adds first valid octal value...
                    // Unless it is not a valid digit, this is only valable for the first digit, maybe wrong
                    if (lastNumericPos != 0)
                    {
                        string startStr = item.Substring(0, lastNumericPos);
                        if (IsNumeric(item.Substring(0, 1)) == -1)
                            bytes.Add((byte)Convert.ToInt32(startStr, 8));
                    }

                    // ...then splits rest into UTF8 chars
                    for (int j = lastNumericPos; j < item.Length; j++)
                    {
                        bytes.Add((byte)Encoding.GetBytes(item.Substring(j, 1).ToCharArray())[0]);
                    }

                }
               
            }

            return bytes.ToArray();
        }

The IsNumeric function that returns -1 if argument is only digits, otherwise returns the index (zero base) of the first non digital character

```
public static int IsNumeric(this string s)
        {
            foreach (char c in s)
            {
                if (!char.IsDigit(c))
                {
                    return s.IndexOf(c);
                }
            }

            return -1;
        }
```
The rest should be pretty obvious but I will be back for solving any further trouble

olikraus, 2019-06-20:
ok.... closing this issue for now...

closed 2019-06-20 by olikraus