olikraus/u8g2 #905: 128x128 using only 64 pixels in Y

mik9900, 2019-05-30:
Hi there, first of all thanks for the library it is really useful.

I'm using the NodeMCU Amica with this oled screen : https://www.banggood.com/1_5-Inch-128x128-OLED-Shield-Screen-Module-For-Raspberry-Pi-STM32-Arduino-p-1305186.html?rmmds=myorder&cur_warehouse=CN

I'm using this constructor : "U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ /*SCL, /* data=*/ /*SDA, /* reset=*/ U8X8_PIN_NONE);"

I can only seem to get 64 pixels in the Y axis. Look at this picture : https://imgur.com/J78U337

Thank you ! 💃 
[image]

olikraus, 2019-05-30:
F***? Please use proper language.

mik9900, 2019-05-30:
> F***? Please use proper language.

Woops didn't realize it was there sorry.

olikraus, 2019-05-30:
Did test the display also with any other constructor?
There is also 
`U8G2_SSD1327_EA_W128128_1_HW_I2C`
and
`U8G2_SSD1327_WS_128X128_1_HW_I2C`

How will these constructors look like? Any improvement?

mik9900, 2019-05-30:
> Did test the display also with any other constructor?
> There is also
> `U8G2_SSD1327_EA_W128128_1_HW_I2C`
> and
> `U8G2_SSD1327_WS_128X128_1_HW_I2C`
> 
> How will these constructors look like? Any improvement?
> 
> How will it look like, if you draw a box at the upper end of 10x10 pixel size?

Thanks for the quick answer.

The first constructor didn't help for my 1/2 pixel but it even made it worse, I don't have pixels at the top nor bottom now : 
[image]

The second constructor doesn't work for me, what am I doing wrong ? :
[image]

If I try the 0, 0, 10 ,10 here is what it looks like :
U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
[image]
I did say 0, 0, 10, 10 but as you can see it's not exactly in the corner...

U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
[image]
Using this constructor, I get the corner but still 1/2 in Y axis.

Thanks again :)

olikraus, 2019-05-30:
> The second constructor doesn't work for me, what am I doing wrong ?

Probably nothing. Either your display is broken or you have revealed a new display with is not yet implemented in u8g2.

Let's modify the MIDAS init sequence a little bit. Please locate u8x8_d_ssd1327.c on your local driver and modify line 491:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L491
to: 
`  U8X8_CA(0x0a0, 0x011),		/* remap configuration */`

Will this help? How will the box or any other text look like?

mik9900, 2019-05-31:
> Probably nothing. Either your display is broken or you have revealed a new display with is not yet implemented in u8g2.

Mmmh I've seen a couple of people talking about the same exact screen that I bought on this github... 
https://www.banggood.com/1_5-Inch-128x128-OLED-Shield-Screen-Module-For-Raspberry-Pi-STM32-Arduino-p-1305186.html?rmmds=myorder

Alright, I modified the u8x8_d_ssd1327.c and tried to write a box 0, 0, 25, 25 + Write "Thanks" with different constructors, here are the results : 

```
#include <Arduino.h>
#include <U8g2lib.h>

// screen oled i2c  128x128 Declaration
U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ /*SCL, /* data=*/ /*SDA, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   //Worse, still 1/2 pixel in Y BUT, nothing at all at the top and bottom.
//U8G2_SSD1327_WS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.setFontMode(1);
    //u8g2.setDrawColor(1);
    u8g2.drawBox(0, 0, 25, 25);
    //u8g2.drawFrame(15, 67, 80, 23);
    u8g2.setFont(u8g2_font_t0_11b_tf);
    u8g2.setCursor(20, 85);
    u8g2.print("Thanks");
  } while ( u8g2.nextPage() );
    delay(1000);
}
```

U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ /*SCL, /* data=*/ /*SDA, /* reset=*/ U8X8_PIN_NONE);
[image]

U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
EXACT SAME THING, The X,Y position are all weird (0,0 box is out of screen and thanks in way too high for a Y85.)

U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
[image]

Seems like the top and bottom parts are compressed still

olikraus, 2019-05-31:
So, with `U8G2_SSD1327_MIDAS_128X128_1_HW_I2C` and `U8X8_CA(0x0a0, 0x011)` the 10x10 box is not visible any more?

Is there any code which works successfully for this display? Is there any description from the supplier?

mik9900, 2019-06-03:
Sorry for the delay, I don't have access to the internet on weekends. 

Yes, using the U8G2_SSD1327_MIDAS_128X128_1_HW_I2C constructor after modifying U8X8_CA(0x0a0, 0x011), You can see that the "thanks" is much higher up than the other constructor while keeping the same code. Thus making the 10x10 box out of the screen... 

I didn't find any code that would make all the pixels usable...  I think I am just going to purchase a real screen instead of this cheap Chinese one.

olikraus, 2019-06-03:
> And no, there aren't much information from the seller which is pretty frustrating.

Well, yes, it is also fun to do this reverse engineering for an unknown display. But yes, it is also frustrating, if you want to build a project, but forced to deal with one part only for weeks.

It is like this: The display (OLED matrix) is connected with the SSD1327 in a specific way requires a specific timing for each pixel and a specific voltage. All this has to be known and according to this, we have to programm the SSD1327. The init sequence, which is here for the MIDAS OLED is here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L477
It just tells the SSD1327 about how to drive the OLED itself. Without this knowledge, we have to guess and reverse engineer the connection.

I personally would do this, but because I do not have the OLED in my lab, you only can do the reverse engineering. I think it is not that complicated, but you need to understand the available SSD1327 configuration commands and change the arguments accordingly.

The SSD1327 datasheet is there on the internet. It will tell you the commands and the possible arguments for them. Then you can start changing the arguments, until you see improvements.

A good candidate in your case is the 0x0a0 command. So we need to change the argument here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L491

You have tested with 0x011, but I think there might be some other combinations for testing.
Also this command could be modified:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L486

There might be even more commands, which are not part of the current init sequence because I didn't deviate from the reset defaults. However all commands are mentioned in the SSD1327 datasheet.

 So to say: We have just started with the reverse engineering process. 
But yes, if you want to finish your project, it is probably better to choose a display which is already supported by u8g2.

olikraus, 2019-06-20:
How to continue here? 

Hmm... is this the same display? https://www.aliexpress.com/item/33009555324.html

Maybe I can order this display...

olikraus, 2019-06-20:
Hmm, the display above does not offer paypal.
Maybe this is also the same: https://www.aliexpress.com/item/32929973938.html

olikraus, 2019-06-20:
I have ordered this display...

mik9900, 2019-06-20:
Ohh sorry I didn't realise that I didn't answer you! 

I decided to order a new LCD that looks a litle better.  To answer your question, yes the one you showed me on aliexpress is the same one that I have.

Thanks again for your help!

olikraus, 2019-06-20:
Ok, great. Thanks for the feedback.

dilucide, 2019-08-08:
I'd argue this issue not to lay within the library but with the particular display of mik9900. Having a look at the photos posted, damage to the left lower corner of the display is rather evident.
It is a common error condition with the SSD* OLEDs that, if the lower corners are damaged, every other pixel of the display does not work.

olikraus, 2019-08-09:
I have received the display from aliexpress. It works with the "U8G2_SSD1327_MIDAS_128X128_1_HW_I2C" constructor without problems.
The only issue is the speed, which defaults to 100KHz for this controller for I2C.
It looks like, that 400KHz is supported for this OLED, so calling "u8g2.setBusClock(400000);" in setup() procedure will increase speed a lot.

olikraus, 2019-08-09:
As a conclusion, I also assume, that your OLED is broken.

closed 2019-08-09 by olikraus

jackthese, 2020-02-08:
Guys... Thanks a million for working all this out for me. I bought a similar (maybe identical) display on Amazon (https://www.amazon.com/gp/product/B07FHLFQWB/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1) and could only get 96 pixels working vertically, using a NodeMCU-32S controller. Three nuances: Had to go to the "F_HW" version to get the buffer big enough to work, did NOT use the HW I2C pins on the ELS32 (too fast?) and ran the clock speed to 1,000,000 with no problem. Very fast. For the record, here's my sketch and screen shot...
`#include <U8g2lib.h>
U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(  // note "F_HW" version gave best results
  /* No Rotation*/U8G2_R0,
  /* reset=*/ U8X8_PIN_NONE,
  /* clock=*/ 25,                          // use NON-default I2C pins for NodeMCU-32S
  /* data=*/ 26);                          // data pin
void setup() {
  u8g2.begin();
  u8g2.setBusClock(1000000);               // seems to work reliably, and very fast
  u8g2.drawLine(0, 0, 127, 127);           // draw edges of screen, and X
  u8g2.drawLine(0, 127, 127, 0);
  u8g2.drawLine(0, 0, 127, 0);
  u8g2.drawLine(0, 0, 0, 127);
  u8g2.drawLine(127, 0, 127, 127);
  u8g2.drawLine(0, 127, 127, 127);
  u8g2.sendBuffer();
}
void loop() {
}`
[image]

jackthese, 2020-02-08:
Sorry about that. Cut and paste into the code box lost the line feeds somehow. Hear's just a text listing...

#include <U8g2lib.h>
U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(  // note "F_HW" version gave best results
  /* No Rotation*/U8G2_R0,
  /* reset=*/ U8X8_PIN_NONE,
  /* clock=*/ 25,                          // use NON-default I2C pins for NodeMCU-32S
  /* data=*/ 26);                          // data pin
void setup() {
  u8g2.begin();
  u8g2.setBusClock(1000000);               // seems to work reliably, and very fast
  u8g2.drawLine(0, 0, 127, 127);           // draw edges of screen, and X
  u8g2.drawLine(0, 127, 127, 0);
  u8g2.drawLine(0, 0, 127, 0);
  u8g2.drawLine(0, 0, 0, 127);
  u8g2.drawLine(127, 0, 127, 127);
  u8g2.drawLine(0, 127, 127, 127);
  u8g2.sendBuffer();
}
void loop() {
}

jackthese, 2020-02-08:
And sorry again: cut and paste lost some *s in lines 5 and 6. All I did was select and copy the code from the Arduino IDE and paste it in as text in this box. I looked in the wrong place for the Preview button, but I'll be sure to use it next time. Note the slope of the slash symbols changed too. Weird....

olikraus, 2020-02-09:
I have fixed the above code Thanks for the feedback!