olikraus/u8g2 #906: Check for U8G2_16BIT on 240px+ displays?

Bzzz, 2019-05-30:
Hi,
quick feature request: Would it be possible to throw some warning when using displays beyond 240px width while U8G2_16BIT isn't enabled? getDisplayWidth clearly knows it's limited, maybe this can be compared to the display data from the clib dir?

I can only assume there's a technical reason this flag isn't set automatically, so that would help with people that do not RTFM properly or worse, only get the idea when opening a new issue. I just spent a good 90 minutes going down several rabbit holes, as updateDisplay() / sendBuffer() would work for a ported old project (Micro + SSD1306 + Adafruit lib -> ESP32 + SSD1322 + your fantastic library), but updateDisplayArea() showed weirdly offset rows/tiles. After drawing 45° lines to confirm that this new display isn't defective, I finally printed getDisplayWidth(), which showed 240, while glitches during flashing used the entire 256px. This led to the discovery of #885, which in return led me to **/clib**/u8g2.h, which led to a fully working 256px display (that I wouldn't have used past 240px for quite some time), which fixed the updateDisplayArea problem. Great fun, learned something, but 1/10 wouldn't recommend to any newbie starting with a nice big display... ;)

olikraus, 2019-05-31:
> I can only assume there's a technical reason this flag isn't set automatically

Indeed it is a technical reason. The dimension of your screen is only known at runtime, while the 16 bit flag has to be set earlier at compile time.

Once you have selected the constructor, we can not jump back in time to enable the flag.

olikraus, 2019-06-02:
hope it is ok, to close this...

closed 2019-06-02 by olikraus