olikraus/u8g2 #907: Sudden flips/mirroring

mkovero, 2019-06-02:
Hi, I've encountered sudden screen flip or sudden mirror on both SSD1309 and SSH1106-based displays using HW SPI and teensy 3.5, works again after recycling power. Do you have any tips where to start poking from or what to do to get displays stable?

olikraus, 2019-06-02:
Will sending out a u8x8.begin() or u8g2.begin() undo the flip?

I remember, that there was a similar discussion somewhere (probably here in one of the issues)...

mkovero, 2019-06-02:
I see similar issue in https://github.com/olikraus/u8g2/issues/760
however, units show this behaviour regardless being connected to pc via usb and they have their own external power source that is monitored and able to provide required current to the displays.
environment itself is pretty hostile signalwise, so I could just try to reset the display every now and then if it helps.

olikraus, 2019-06-03:
It might be sufficient to call https://github.com/olikraus/u8g2/wiki/u8g2reference#initdisplay from time to time.

olikraus, 2019-06-20:
Can we close this?

closed 2019-10-03 by olikraus