olikraus/u8g2 #908: U8G2 - I2C Multiplexer

JoseAllgaeu, 2019-06-02:
Hello,

I'm using one SSD1327 OLED 128 x 128 connected via I2C.
Using this Constructor based on Issue #880, Running perfectly:

```
U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```

I would like to connect a second Display:

```
// OLED A:
U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2A(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// OLED B:
U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2B(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```

It seem to me, the Default I2C Adress 0x3C of the Display is not changeable.

Is it possible to use this I2C Multiplexer in conjunction with two SSD1327 Displays: https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout?view=all 

Based on Issue #434:

```
// OLED A
u8g2A.setI2CAddress(0x3C * 2);
// OLED B
u8g2B.setI2CAddress(0xXY * 2);
```
XY = I2C Adress given by the Multiplexer

Many thanks in advanced!

olikraus, 2019-06-02:
It is possible, but it will not be that easy with the TCA9548. I suggest you should read more about the TCA9548: It acts as a third I2C slave which has to be programmed manually by you.
This means, you have to keep both displays at the same address (0x3c). Let me assume, you can select display A with tca9548dispA() and display with tca9548dispB(), then any u8g2 command which issues any data to the display has to be prefixed with this select statement:
```
tca9548dispA(); 
u8g2A.begin();
tca9548dispB(); 
u8g2B.begin();
```
Actually, you do not even need to use the object twice, you can work with one u8g2 object only in your case:
```
tca9548dispA(); 
u8g2.begin();
tca9548dispB(); 
u8g2.begin();
```

JoseAllgaeu, 2019-06-04:
Hi Olli,

many thanks for your fast response. I'll keep you informed.

closed 2019-06-04 by JoseAllgaeu

JoseAllgaeu, 2019-06-04:
Hi Olli,

I solved it through soldering the Resistor to an other Position to get a different I2C Adress. Followed based on Issue #434.

olikraus, 2019-06-04:
Sounds good.