olikraus/u8g2 #912: Seeed U8g2

LynnL4, 2019-06-04:
I noticed that you used hardware I2C to drive OLED on Seeeduino, but I couldn't work properly when I tested it. There was no display, and it worked well when connecting two OLED displays  to Seeeduino. The software I2C is work well. What could be the reason? Thanks!

LynnL4, 2019-06-04:
I hope U8g2 can support Kendryte K210, but Kendryte K210 does not support Hardware SPI yet。Thanks

olikraus, 2019-06-04:
What exactly is your setup? Which display did you use?

LynnL4, 2019-06-05:
I use the seeed_128X128 display, and use the following setup.
U8G2_SH1107_SEEED_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2019-06-06:
Strange. for me, exactly the same setup works without problems:
https://github.com/olikraus/u8g2/wiki/gallery#22-apr-2019--sh1107-128x128-seeed-studio-oled

olikraus, 2019-06-20:
Any further comments here? Any other way how I can support here?

LynnL4, 2019-06-20:
It is working normally now. There may be something wrong with my board. Thank you for your attention.

closed 2019-06-20 by LynnL4

olikraus, 2019-06-20:
:)