olikraus/u8g2 #917: How to make a font list detail (image of all chars) ?

FrenchLab47, 2019-06-07:
Hello,

Following your advice, I managed to change accents, or add glyphs in the fonts I want to use.
Now I would need to build a detailed list or picture of all chars, with the ones I added, like on your detail pages.
[https://github.com/olikraus/u8g2/wiki/fntgrpu8g](url)

Can you please tell me, with which software did you make these images?
Thank you for your help and your work.

olikraus, 2019-06-07:
You can generated a (different) overview picture with bdfconv itself by using the option "-d".
"-d" expects a bdf file as argument to render the information text. So you need to provide another bdf file (could be the same, but must included lower and upper chars).
bdfconv will then generate a .TGA file with a fixed name in the current directory.
There are also some more options to modify this picture.

The actual pictures, which you can see on the wiki pages, are generated by a different tool: The u8g2 font release build procedure https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c
Unfortunately this can not be used for individual fonts.

FrenchLab47, 2019-06-07:
Yes !
Thank you for this effective and instant response.
I just tested and that's exactly what I need.
Best regard.

olikraus, 2019-06-07:
:)

closed 2019-06-20 by olikraus