olikraus/u8g2 #919: Can the library add 12832_ssd1316 driver support?

iggymayer, 2019-06-10:


olikraus, 2019-06-10:
Any datasheet?

iggymayer, 2019-06-10:
https://mega.nz/#!ZYciEArY!Lo3eSw-zPivsn8vh0SdPWwxxSn3IOOflY8g9t9ZRwYo

iggymayer, 2019-06-10:
https://mega.nz/#!ZR1j2AQK!zhKiLkmi3bD4wjMEKTztI6nId7er3F2o7T0uoK6jTmU
QT-2832TSWUG02

iggymayer, 2019-06-10:
https://mega.nz/#!pUEmVQ7S!oMuezOppe8Rn8JhX0E22iPJRMJIz_6gIBViu-f1LSaI
QT-2832TSWZG01 VER A

olikraus, 2019-06-12:
I have created beta 2.26.12 which supports SSD1316.
Constructors look like this: 
```
U8G2_SSD1316_128X32_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```
Can you test the page bufferflip mode example? There should be a frame around the display all the time?
Will it work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

iggymayer, 2019-06-13:
Thank you very much. I will test when the screen is mailed to me.

iggymayer, 2019-06-18:
Hello, now the screen is lit (QT-2832TSWZG01 VER A), but the displayed font is mirrored, I need to use "U8G2_MIRROR" to display properly, I hope you can adjust it, thank you.

iggymayer, 2019-06-18:
ESP32-DevKitC cannot light the screen.
Example "Hellow World"
U8G2_SSD1316_128X32_F_4W_SW_SPI u8g2 (U8G2_MIRROR, 14, 13, 32, 33, 35);

olikraus, 2019-06-18:
Can you sent a picture of the mirrored version?
`U8G2_SSD1316_128X32_F_4W_SW_SPI u8g2 (U8G2_R0, 14, 13, 32, 33, 35);`
Can you located u8x8_d_ssd1316.c on your local drive. Then modify line 166
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1316.c#L166

from `U8X8_C(0x0c8)` to `U8X8_C(0x0c0)`

Will this help?

iggymayer, 2019-06-18:
arduino uno & U8G2_R0
[https://mega.nz/#!MIVBQA4A!ze2YuLmTI54agJUQqHGdLuaGREBZAgo0nV4aw17vpVs](url)

iggymayer, 2019-06-18:
arduino uno & U8G2_R0
https://mega.nz/#!MIVBQA4A!ze2YuLmTI54agJUQqHGdLuaGREBZAgo0nV4aw17vpVs

iggymayer, 2019-06-18:
Line 166 has been modified, but the screen cannot be lit with ESP32-DevKitC. Whether it is U8X8 or U8G2, there is no problem with the development board and screen circuit.

iggymayer, 2019-06-18:
Sorry, I just read it wrong. According to your method, the mirroring problem has been solved and can be displayed normally under U8G2_R0.

olikraus, 2019-06-18:
So, changing U8X8_C(0x0c8) to U8X8_C(0x0c0) will solve the problem?

iggymayer, 2019-06-18:
The mirroring problem has been resolved, the connection arduino can be displayed normally, but nothing is displayed on the ESP32.

olikraus, 2019-06-20:
ok, i fixed the code for the mirror problem.
For the ESP32 issue: It should work. Is your wiring correct?

olikraus, 2019-06-20:
I will close this issue. Please reopen if there are any issues regarding the SSD1316 implementation. 
Please create a new issue to discuss the ESP32 problem (but probably I need more information on the wiring).

closed 2019-06-20 by olikraus