olikraus/u8g2 #920: Wrong argument to SPI command 0x44 for IL3820

amanita-citrina, 2019-06-11:
Array u8x8_d_il3820_296x128_init_seq[] (and also u8x8_d_il3820_v2_296x128_init_seq[]) in file <code>u8x8_d_il3820_296x128.c</code> contains the line <code> U8X8_CAA(0x44, 0, 29), /* RAM x start & end, 32*4=128 */</code> to set the RAM X end address.

According to the IL3820 datasheet, 29 is the default value (for a display with an X size of 240 pixels). The correct value for the 128x296 display should be (128/8)-1=15.

(In other words, the X dimension spans 16 bytes from 0x0 to 0xF, which amounts to 128 pixels.)

olikraus, 2019-06-12:
Thanks for the feedback.

You mean this line? https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_il3820_296x128.c#L298

so it should be `U8X8_CAA(0x44, 0, 15)`?

You are probably right, but will this make a visible difference? Did you test the result? What exactly is the improvement?

amanita-citrina, 2019-06-12:
Yes, that's the line. I am currently writing my own implementation of the 128x296 display driver, based on your work, the code by Waveshare and the code by ZinggJM.

So far, I have no working code - just reading and (slowly) implementing. The corrected version will have no visual difference. In my understanding, it will allow to write a complete image (16x296 bytes) without rewriting the cursor (command 0x4E/0x4F).

See line 224 of https://github.com/waveshare/e-Paper/blob/master/2.9inch_e-paper_code/RaspberryPi/bcm2835/obj/EPD_2in9.c - they also use a wrong value for command 0x44.

olikraus, 2019-06-12:
ah, i see
the current u8g2 implementation works like this (pseudo code):
with each page if the display
   position the cursor at the beginning of the page
   write the page data

so there is no need for wrapping at the end of a page in the u8g2 code...

olikraus, 2019-06-20:
I have added a comment in the code, but i didn't change the code itself. I think the fix is not required in my case.

closed 2019-06-20 by olikraus