olikraus/u8g2 #921: Support for ST75320

olikraus, 2019-06-12:
Display:
https://www.aliexpress.com/item/4-0-320X240-320240-Graphic-LCD-Module-Display-Panel-Screen-LCM-with-ST75320-Controller-Gray-LCD/33015590137.html

[image]

ST75320: https://www.newhavendisplay.com/appnotes/datasheets/LCDs/ST75320.pdf

closed 2019-06-15 by olikraus

pat1, 2020-06-19:
I apologize for this slightly off topic support request.
I bought an ST75320 based display and in particular http://www.jlxlcd.cn/html/zh-detail-867.html
https://it.aliexpress.com/item/2055196707.html  (3.3V I2C version)

I would like to use it with I2C but I'm not sure of the electrical connections.
I tried:
* D0 -> SCL
* D1 D2 D3 D4 D5 (connected) -> SDA
* D6 -> GND
* D7 -> GND
* VDD -> 3.3V
* VSS -> GND

but nothing responds to an I2C scan.
The datasheet is in oriental language which is not comprehensible to me.

Thanks in advance for any suggestion
Paolo

olikraus, 2020-06-21:
I am happy to see, that someone uses this type of display. 
From the ST75320 i can confirm this:
 * D0: SCL
 * D1, D2, D3, D4, D5 must be tied together and are SDA
 * D6, D7 are the two low bits of the device address bits

Additionally you should connect A0 and CS to high (as adviced in the controller datasheet).
Finally you must connect the reset input to the display somehow: Connecting RES to high might work, but maybe better let u8g2 control the RES (reset) signal: This means, connect the RES input of the display to any empty GPIO of your controller and pass the GPIO number to the u8g2 constructor.

pat1, 2020-06-22:
Thanks a lot !
On my display PCB I suppose A0 is named RS
Finally this works for me:
*   D0 -> SCL
*   D1 D2 D3  (connected) -> SDA
*   D6 D7 -> VDD/VSS   (the two low bits of the device address bits, I2C address fom 60 to 63)
*   RS -> VDD
*   CS -> VDD
*   RES  -> VDD  ( or better connect the RES input of the display to any empty GPIO of your controller and pass the GPIO number to the u8g2 constructor)
*   VDD -> 3.3V
*   VSS -> GND
*   LEDA -> 3.3V
and everything works well ! 
I need to adjust contrast with u8g2.setContrast(80) but the contrast is not very good; is possible it's due to 3.3V suppply?

Thanks for your support and your very good library !
Paolo

olikraus, 2020-06-22:
Remember to put u8g2 into 16 bit Mode (see FAQ). 
For the contrast: maybe a 1uF cap between Gnd and 3.3v very close to the display pcb may help. In general the JLX displays are not that good regarding contrast.

BarbaraTom94, 2020-07-27:
Hi!! I apologize for the off topic support request but I have a problem with one of these displays too. 
I bought an ST75320 based display, in particular: https://www.alibaba.com/product-detail/3-8-94-7-86-7MM_62256129811.html?spm=a2700.12243863.0.0.4ceb3e5fKtjcrS (3.3V with PCB). 
Actually I thought it had i2c interface but it seems that only the version without PCB can have it. Since I have the version with PCB, It seems I can just use the 8-Bit Parallel or serial SPI interface. I'm trying with SPI interface but I'm not sure about electrical connections. 
I just have this datasheet 
https://us-icbuim-file.oss-us-east-1.aliyuncs.com/file/e049280b752e1e9e04eabead639b99de.pdf?Expires=1595946409&OSSAccessKeyId=LTAIxlzvmECyZvmX&Signature=%2BQppaLd1zJaS052Fz8s%2BZ5cpu90%3D&response-content-disposition=fileName%3DSLG320240G.pdf
Thanks in advance for any suggestion.
Barbara

olikraus, 2020-07-27:
Additionally you need the ST75320 datasheet. Section 7.1 in this datasheet explains details of the SPI, I2C (and parallel) interface. For example D0 is the SPI Clock line if the interface is configured as SPI.

First step should be a mapping table as mentioned above.

BarbaraTom94, 2020-08-03:
Thank you very much, I did it!!