olikraus/u8g2 #922: HX1230 hardware SPI problem

jordisalichs, 2019-06-15:
Hi,
I'm using HX1230 display with an Arduino zero board. By default Arduino SPI not support 9 bits, by this way I have to use software SPI, but it is very slow. It needs about 100ms to update display. I need to work faster. HX1230 can work SPI 4Mbits.
 I modificate Arduino in order to support 9 bits hardware SPI, with logic analyser I checked that it works fine. 
[image]
Then I modificate u8g2 library in order to work with Arduino Zero hardware SPI. I add image from logic analyser from software (running fine, but slow) and hardware SPI.
Software, only one reset (Channel 7):
[image]
[image]
Hardware (Several resets and not all bytes are sent.:
[image]
[image]

I do not know how to continue. Can somebody help me?

Thank you

olikraus, 2019-06-15:
What is the actual problem?

jordisalichs, 2019-06-15:
Display does not show anything with modified library. 
Problems: 
1. Several reset (Channel 7 last image).
2. I'm not 100% sure, the same code send less bytes.
I just make the same strategy with https://github.com/bitbank2/bb_hx1230 library and display works fine and it is updated in less than 8ms.SPI at 4Mbps. But library is very limited.

jordisalichs, 2019-06-15:
More test...
I changed reset PIN in constructor and make reset manually.
Now display shows only the first 8 pixels lines. (update time is 4ms) SPI working at 4Mbits/s.
Using hardware SPI only send 1.001 bytes. First bytes NOT send are:
0x000, 0x010, 0x0B1 after a lot of 0x100.
I tested Hardware SPI with logic analyser sending 4096 bytes without problems,
Is it possible buffer limit in library?

olikraus, 2019-06-16:
> Is it possible buffer limit in library?

Did you use the full buffer setup? It sound you have used the sendBuffer together with the page mode constructor. This would be wrong. The sendBuffer must be used only with the full buffer constructor. Maybe you can post your code.

jordisalichs, 2019-06-16:
This is the code using Hardware SPI contructor, that only shows the first 8 pixels lines.
-------------------------------
U8G2_HX1230_96X68_1_2ND_4W_HW_SPI u8g2 = U8G2_HX1230_96X68_1_2ND_4W_HW_SPI(U8G2_R0, 31, 33); // 30
-------------------------------
  pinMode(30, OUTPUT);
  digitalWrite(30, LOW);
  delay(50);
  pinMode(30, OUTPUT);
  digitalWrite(30, HIGH);

  u8g2.begin();

  u8g2.setContrast(128);
  
  long before;
  long after;
  long time = 0;
  char line[25];
  long i = 0;
    
  while(true)
  { 
    before = millis();
    u8g2.clearBuffer();          // tested with and without this line
    u8g2.setFont(u8g2_font_profont11_mr);

    sprintf(line, "C: %d T: %d", i, time);
    u8g2.drawStr(0, 9, line);

    u8g2.drawStr(0, 20, "123456789");
    sprintf(line, "C: %d T: %d", i, time);
    u8g2.drawStr(0, 30, line);
    sprintf(line, "C: %d T: %d", i, time);
    u8g2.drawStr(0, 40, line);
    sprintf(line, "C: %d T: %d", i, time);
    u8g2.drawStr(0, 50, line);
    sprintf(line, "C: %d T: %d", i, time);
    u8g2.drawStr(0, 60, line);
  
    u8g2.sendBuffer();          // transfer internal memory to the display
    after = millis();
    time = after - before;

    i++;
  }

------------------------------
If I use software SPI, this code works fine. 125ms to update the full display.
U8G2_HX1230_96X68_F_3W_SW_SPI u8g2 = U8G2_HX1230_96X68_F_3W_SW_SPI(U8G2_R0, /* clock=*/ 25, /* data=*/ 24, /* cs=*/ 31, /* reset=*/ 30);  
----------------------------

jordisalichs, 2019-06-16:
This is the code using Hardware SPI contructor, that only shows the first 8 pixels lines.

U8G2_HX1230_96X68_1_2ND_4W_HW_SPI u8g2 = U8G2_HX1230_96X68_1_2ND_4W_HW_SPI(U8G2_R0, 31, 33); // 30

pinMode(30, OUTPUT);
digitalWrite(30, LOW);
delay(50);
pinMode(30, OUTPUT);
digitalWrite(30, HIGH);

u8g2.begin();

u8g2.setContrast(128);

long before;
long after;
long time = 0;
char line[25];
long i = 0;

while(true)
{
before = millis();
u8g2.clearBuffer(); // tested with and without this line
u8g2.setFont(u8g2_font_profont11_mr);

sprintf(line, "C: %d T: %d", i, time);
u8g2.drawStr(0, 9, line);

u8g2.drawStr(0, 20, "123456789");
sprintf(line, "C: %d T: %d", i, time);
u8g2.drawStr(0, 30, line);
sprintf(line, "C: %d T: %d", i, time);
u8g2.drawStr(0, 40, line);
sprintf(line, "C: %d T: %d", i, time);
u8g2.drawStr(0, 50, line);
sprintf(line, "C: %d T: %d", i, time);
u8g2.drawStr(0, 60, line);

u8g2.sendBuffer();          // transfer internal memory to the display
after = millis();
time = after - before;

i++;
}
--
If I use software SPI, this code works fine. 125ms to update the full display.
U8G2_HX1230_96X68_F_3W_SW_SPI u8g2 = U8G2_HX1230_96X68_F_3W_SW_SPI(U8G2_R0, /* clock=/ 25, / data=/ 24, / cs=/ 31, / reset=*/ 30);

jordisalichs, 2019-06-16:
Yes it is working!
The correct constructor is:
U8G2_HX1230_96X68_F_2ND_4W_HW_SPI u8g2 = U8G2_HX1230_96X68_F_2ND_4W_HW_SPI(U8G2_R0, 31, 33); // 30

olikraus, 2019-06-16:
:)

jordisalichs, 2019-06-16:
Thank you for your help. 
My application speeds:
- Software SPI needs between 100 and 121ms
- Hardware SPI needs between 6 and 26ms
- Hardware SPI (not wait after write) needs between 5 and 23ms

closed 2019-06-20 by olikraus