olikraus/u8g2 #924: SSD1306 dual I2C display -

fm4dd, 2019-06-16:


fm4dd, 2019-06-16:
I am running two SSD1306 128x64 OLEDs on a single I2C bus with separate addresses (0x3C and 0x3D). They work as listed in issue [#434](https://github.com/olikraus/u8g2/issues/434).

However unless the buffer fully cleared  after writing to one display, updating them independently does not work. I believe the buffer pointers are both written into, but let me first explain the issue on a few examples.

Example 1:
`#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_F_HW_I2C dleft(U8G2_R0);
U8G2_SSD1306_128X64_NONAME_F_HW_I2C dright(U8G2_R0);

void setup() {
  dleft.setI2CAddress(0x3C * 2);
  dright.setI2CAddress(0x3D * 2);
  dleft.begin();
  dright.begin();
  dleft.clearBuffer();
  dright.clearBuffer();
}

void loop() {
  // write ABC to left display
  dleft.clearBuffer();
  dleft.setFont(u8g2_font_inr19_mr);
  dleft.drawStr(0,32, "ABC");
  dleft.updateDisplay();
  
  // write DEF to right display
  //dright.clearBuffer();
  dright.setFont(u8g2_font_inr19_mr);
  dright.drawStr(64,32, "DEF");
  dright.updateDisplay();
  
  delay(500000);
}`

In this case, the intention is to write ABC to display-1 (left, I2C 0x3C), and DEF to display-2 (right, I2C 0x3D).  It only works after I uncomment the dright.clearBuffer() line, but that should not be necessary. The right display gets the content from the left display as well, even though that wasn't intended.

closed 2019-06-16 by fm4dd

olikraus, 2019-06-19:
Yes, the memory is shared between the displays.