olikraus/u8g2 #926: Support for SPD0301

liamlacey, 2019-06-17:
I am currently testing the library with the RiTdisplay P30804 OLED module that uses the SPD0301 controller chip.

I am currently using the u8g2_Setup_ssd1309_128x64_noname0 structure for this module using SPI comms and everything appears to be working/displaying correctly, however is there likely to be any low level settings that need adjusting for this controller chip or OLED module?

The SPD0301 datasheet can be found at https://www.usmicroproducts.com/sites/default/files/SPD0301_1.0.pdf.

Many thanks.

olikraus, 2019-06-19:
Not sure how to continue here. The mentioned datasheet is for the controller. The low level settings (if any) are usually mentioned in the display module datasheet of the OLED.

If it works, I do not know what I can do better...

liamlacey, 2019-07-01:
I assumed as much - just that verification helps.
Thanks!

closed 2019-07-01 by liamlacey