olikraus/u8g2 #927: Convert font profont12.

hevet, 2019-06-17:
Hi. In the profont12 font, I changed the dot so that, for example, the temperature could be saved more or less 24.45 and not 24. 45. You would convert it to be used in u8g2 and add it in the program code.
[ProFont12v2.zip](https://github.com/olikraus/u8g2/files/3296426/ProFont12v2.zip)

olikraus, 2019-06-17:
ok, but I assume, that this will only affect the tr font and not the mr font.
Or shell I introduce profont12v2 as a new tr font?
I think this might be better, otherweise it might break existing arduino code.

hevet, 2019-06-17:
Only need to profont12v2_tr.

olikraus, 2019-06-18:
Just for clarification: You want me to add profont12v2 to the u8g2 repo or you just need profont12v2 for your personal project? 
I mean, you could simply create the c array by yourself... https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L216

hevet, 2019-06-18:
I need for my personal project. And you could generate the correct file for me, I generated it, but I do not know how to add it to the program code correctly, once it has not seen it at all, another program was still some other error.

olikraus, 2019-06-20:
Is this the font you need?
You can just copy&paste the code into your ino file:
```
const uint8_t u8g2_font_profont12v2_tr[980] U8G2_FONT_SECTION("u8g2_font_profont12v2_tr") = 
  "`\0\3\2\3\4\3\4\4\6\13\0\376\10\376\11\377\1C\2\212\3\267 \5\0\242\3!\6A\243"
  "\343\22\42\7\233\272#\261\4#\13-\256\247\62(\225A\251\0$\17U\236+[*J\266%Q\242"
  "T\266\10%\16E\242gP\22\245\233\244(\211\224\0&\15E\242G\252\324*Q\222H\221\22'\5"
  "\31\273c(\10\323\236+)\365V)\11\323\236#+\365R\2*\12-\256\253RY\232\42\0+\12"
  "-\246+\214\6)\214\0,\10\242\32cH\24\0-\5\213\256c.\6\22\242b\10/\14U\232\263"
  "\26fa\26fa\10\60\15E\242g\311\244%\231\64-Y\0\61\11E\242+\32{\32\4\62\12E"
  "\242g\311\302\254\343 \63\14E\242g\311\302H\15\265d\1\64\14E\242/\223\222R\62ha\66\65"
  "\14E\242\343\30\16i\250%\13\0\66\14E\242g\11\303!\311l\311\2\67\12E\242c\20k\305&"
  "\0\70\14E\242g\311\264d\311l\311\2\71\14E\242g\311l\311\20V\26\0:\10\62\243c\10\207"
  "\0;\11\302\232c\10\207D\1<\7\274\246/jl=\10\35\256cP\7\1>\10\274\246#lj"
  "\3\77\13E\242g\311\302Z\35\212\0@\16E\242g\311\224!I\224!L\207\0A\15E\242+K"
  "\242\244\246\15C\246\5B\15E\242cH\62mP\62\333\240\0C\12E\242g\311\304n\311\2D\12"
  "E\242cH\62\337\6\5E\13E\242\343\30\16IX\34\4F\12E\242\343\30\16I\330\10G\14E"
  "\242g\311\304\322\246%\13\0H\12E\242#\263\15C\346\26I\12E\242c\220\302\236\6\1J\12E"
  "\242\263\243\246%\13\0K\15E\242#\223\222\222\246%Q%\13L\10E\242#\354\343 M\14E\242"
  "#[\226DI\64\267\0N\13E\242#\233\224D\332\334\2O\12E\242g\311|K\26\0P\14E"
  "\242cH\62mP\302F\0Q\12M\236g\311|\251\254\1R\14E\242cH\62mP\62\267\0S"
  "\13E\242g\311\324\65\324\222\5T\11E\242c\220\302>\1U\11E\242#\363[\262\0V\14E\242"
  "#sKJI\26F\0W\12E\242#\363\222(\311-X\13E\242#\263%\265JM\13Y\12E"
  "\242#sKja\11Z\12E\242c\20\263\216\341 [\10R\237c\351\27\1\134\14\325\232#L\303"
  "\64L\303\64\14]\10\322\236C\351\227\1^\10\35\266+Kj\1_\6\16\232\343\0`\6\222\276#"
  "\12a\11-\242g\320L\212\22b\14E\242#,\16If\33\24\0c\11-\242g\311\304t\10d"
  "\12E\242\263e\320l\311\20e\12-\242g\311\206!\35\2f\12\304\242K\211\246\254\23\0g\14="
  "\232g\320l\311\20&\13\0h\12E\242#,\16I\346\26i\11\303\242'\7\244\226\1j\12S\232"
  "+\7\244N\13\0k\13E\242#lJJS%\13l\10\303\242C\352\313\0m\14-\242cH*"
  "J\242$J\1n\11-\242#\61i\266\0o\11-\242g\311l\311\2p\14=\232cH\62\333\240"
  "\204!\0q\12=\232g\320l\311\20\26r\11-\242#\61\211E\0s\11-\242gP\17\12\0t"
  "\11\304\242'+MYQu\11-\242#sR\224\0v\13-\242#\323\222R\222E\0w\14-\242"
  "#I\224DI\224.\0x\12-\242#Kj\225Z\0y\13=\232#sK\206\60Y\0z\11-"
  "\242c\320\332\6\1{\12\333\232+\211Z\262\250-|\6Q\237\343!}\13\333\232#\213\332\222\250%"
  "\2~\6\25\262\307\5\177\5\0\242\3\0\0\0\4\377\377\0";
```

hevet, 2019-06-20:
Thanks, I made a small mistake in the font, but I already know what's going on, I corrected myself and it works. Thanks for your help, you can remove this font from the repository.

olikraus, 2019-06-20:
ok...

closed 2019-06-20 by olikraus