olikraus/u8g2 #928: Very slow rendering

Romancze, 2019-06-18:
Hi 
I'm beginner with arduino so I don't know where is the problem.
I have a project for air quality measurement with ST7920 and arduino nano (also tried ATMega2560). I tried examples for the LCD and that was all right, high FPS. But my code is very slow with your library. I tried different LCD (from nokia) and different library -that was OK.
I tried all choices: HW SPI, SW SPI, paralel mode. All has a same speed. The full buffer is faster but there is an artefact on the screen. 
Each next library and input reading cause slower rendering. 

My video:
SPI page buffer
https://drive.google.com/file/d/1woC1GbwHkCrl3m0ad9YJlitB_gHxcZtZ/view?usp=sharing

SPI full buffer (small artefact on the "T" symbol on the top line after few seconds)
https://drive.google.com/file/d/1VWaos_fm-Bd_ugFqL4zGb3AHvz8l-G62/view?usp=sharing

My code is:

#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>

#include <U8g2lib.h>           //LCD
#include <Adafruit_BMP280.h>  //BMP280 
#include "DHT.h"              //DHT22 
#include <MQ135.h>            //MQ135 CO2

#include <DS3231_Simple.h>    //Clock
#include <SD.h>               //SD card/

#include <SparkFun_VEML6075_Arduino_Library.h> //VEML6075 UV sensor

//////////////////////////////////////
VEML6075 uv;                //UV

DS3231_Simple Clock;        //Clock

U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R3, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 5);    //LCD
//U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R3, /* CS=*/ 10, /* reset=*/ 8);
//U8G2_ST7920_128X64_F_8080 u8g2(U8G2_R3, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 18 /* A4 */, /*cs=*/ U8X8_PIN_NONE, /*dc/rs=*/ 17 /* A3 */, /*reset=*/ 15 /* A1 */);  // Remember to set R/W to 0 

Adafruit_BMP280 bmp;          //BMP280 start I2C
  int korekce = 24;           //BMP280 
                              
                             //Sharp GP2Y1014AU0F
const int sharpLEDPin = 9;   // D9 - LED - White wire.
const int sharpVoPin = A0;   // A0 - Vo - Yellow.
#define USE_AVG              
   #ifdef USE_AVG            
   #define N 200
   static unsigned long VoRawTotal = 0;
   static int VoRawCount = 0;
   #endif 
   static float Voc = 0.6;
   const float K = 0.5;

#define DHTPIN 10               //DHT22 pin10 digital
  #define DHTTYPE DHT22         //DHT22 (AM2302), AM2321 
  DHT dht(DHTPIN, DHTTYPE);     //DHT22 inicializace

#define pinA A2                 //MQ135
  MQ135 senzorMQ = MQ135(pinA);

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////
 void setup(void) {
  Serial.begin(9600);
   Serial.println("setup");

 Clock.begin();                //Clock
  
  bmp.begin(0x76);             //Define i2c BMP280
      bmp.setSampling(Adafruit_BMP280::MODE_NORMAL,     /* Operating Mode. */
                  Adafruit_BMP280::SAMPLING_X4,     /* Temp. oversampling */
                  Adafruit_BMP280::SAMPLING_X16,    /* Pressure oversampling */
                  Adafruit_BMP280::FILTER_X16,      /* Filtering. */
                  Adafruit_BMP280::STANDBY_MS_500); /* Standby time. */
               
  pinMode(sharpLEDPin, OUTPUT); //Sharp GP2Y1014AU0F

   
  u8g2.begin();                 //LCD
    u8g2.enableUTF8Print();
}

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////
void loop(void) {
  Serial.println("loop");

  
////////////
   float tlak = ((bmp.readPressure() + korekce*100.0)/100.0);   
     Serial.println(tlak);
   float tep = bmp.readTemperature();
     Serial.println(tep);
   float vlh = dht.readHumidity();
     Serial.println(vlh);
   float CO2 = senzorMQ.getPPM();
      Serial.println(CO2);
   float O3 = analogRead(A1);
      Serial.println(O3);
   float NO2 = analogRead(A6);
      Serial.println(NO2);
   float CO = analogRead(A7);
      Serial.println(CO);
            

//   Serial.println(String(uv.uva()) + ", " + String(uv.uvb()) + ", " + String(uv.index()));

   
/////////////Clock

  Clock.printDateTo_YMD(Serial);
  Serial.print(' ');
  Clock.printTimeTo_HMS(Serial);
  Serial.println();

/*
/////////////Sharp GP2Y1014AU0F
  digitalWrite(sharpLEDPin, LOW);           
   delayMicroseconds(280);
   int VoRaw = analogRead(sharpVoPin);
   digitalWrite(sharpLEDPin, HIGH);
   delayMicroseconds(9620);
   #ifdef PRINT_RAW_DATA
   printValue("VoRaw", VoRaw, true);
   Serial.println(" Sharp");
   delay(1000);
   #endif 
   float Vo = VoRaw;
   #ifdef USE_AVG
   VoRawTotal += VoRaw;
   VoRawCount++;
    if ( VoRawCount >= N ) {
     Vo = 1.0 * VoRawTotal / N;
     VoRawCount = 0;
     VoRawTotal = 0;
    } else {
    return;
    }
    #endif // USE_AVG

  Vo = Vo / 1024.0 * 5.0;   // Compute the output voltage in Volts.
//  printFValue("Vo", Vo*1000.0, "mV");

  float dV = Vo - Voc;      // Convert to Dust Density in units of ug/m3.
  if ( dV < 0 ) {
    dV = 0;
    Voc = Vo;
  }
  float dustDensity = dV / K * 100.0;
//  printFValue("DustDensity", dustDensity, "ug/m3", true);
  Serial.println("Konec");
  delay(2000);

*/

 ////////////////////////   LCD 

 u8g2.firstPage();  
  do {

    u8g2.setFont(u8g2_font_oskool_tr);
        u8g2.drawStr(0,8,"Kvalita");
        u8g2.drawStr(7,16,"prostredi");
     u8g2.drawLine(0, 17, 64, 17);
        
   // u8g2.setFont(u8g2_font_5x7_mr);
   u8g2.setFont(u8g2_font_5x7_mr);
   

        u8g2.drawStr(29,26,"C");  
             u8g2.drawStr(59,26,"%");
        u8g2.drawStr(0,35,"T:");
        u8g2.drawStr(0,44,"CO ");
        u8g2.drawStr(0,53,"CO ");
        u8g2.drawStr(0,62,"NO ");       
        u8g2.drawStr(0,71,"O  ");
        u8g2.drawStr(0,80,"NH ");
        u8g2.drawStr(0,89,"PM ");
        u8g2.drawStr(0,98,"E ");
        u8g2.drawStr(0,107,"UV ");
        u8g2.drawStr(0,116,"UV  ");

        

    u8g2.setFont(u8g2_font_u8glib_4_hr);
        u8g2.drawStr(25,23,"o"); 
        u8g2.drawStr(10,44,"2"); //CO2
        u8g2.drawStr(10,62,"2"); //no2
        u8g2.drawStr(5,71,"3"); //o3
        u8g2.drawStr(10,80,"3"); //nh3
        u8g2.drawStr(10,89,"10"); //pm10
        u8g2.drawStr(10,107,"A");
        u8g2.drawStr(10,116,"B");
        
        
        u8g2.drawStr(42,44,"ug/m3");
        u8g2.drawStr(42,44,"ug/m3");
        u8g2.drawStr(42,53,"ug/m3");
        u8g2.drawStr(42,62,"ug/m3");
        u8g2.drawStr(42,71,"ug/m3");
        u8g2.drawStr(42,80,"ug/m3");
        u8g2.drawStr(42,89,"ug/m3");
        u8g2.drawStr(45,98,"lux");
        u8g2.drawStr(37,107,"uW/cm2");
        u8g2.drawStr(37,116,"uW/cm2");
   

   u8g2.setFont(u8g2_font_5x7_mr);
     u8g2.drawLine(0, 119, 64, 119);
        u8g2.drawStr(0,128,"Baterie: XXX");
        

     u8g2.setCursor(10,35);
     u8g2.print(tlak, 0);    
     u8g2.setCursor(0,26);
     u8g2.print(tep);  
     u8g2.setCursor(45,26);
     u8g2.print(vlh, 0); 
     u8g2.setCursor(15,44);
     u8g2.print(CO2, 0);  
   
   delay(2000);

  } while( u8g2.nextPage() );
}

olikraus, 2019-06-19:
The "delay(2000)" must be outside the firstPage/nextPage Loop.

olikraus, 2019-08-13:
no further response...

closed 2019-08-13 by olikraus