olikraus/u8g2 #930: U8G2 text back to front

GavinBrownAus, 2019-06-21:
Hi, 

Working with a JLX256160 display in 4 wire spi mode and I notice that the text is displaying back to front. See attached pic. 

Using a Teensy 3.6 and the JLX256160. 

Connections are as follows:

[image]

[image]

Using the following constructor:

U8G2_ST75256_JLX256160_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

16 bit mode is set.

Any help is greatly appreciated.

olikraus, 2019-06-22:
What is the exact display name? Is there any datasheet for your display?

olikraus, 2019-06-22:
Can you try "U8G2_ST75256_JLX256160_ALT_F_4W_SW_SPI" constructor?

GavinBrownAus, 2019-06-22:
See attached datasheet. Original is in Chinese, also attached is the
translated version.

GavinBrownAus, 2019-06-22:
Thanks Oliver for the quick response. Tried the new constructor and now have the following result. 

Text is the right way round, but is offset from the top of the screen and wraps over itself. 

[image]

olikraus, 2019-06-22:
Hmm... I think github has filtered the attachments.

olikraus, 2019-06-22:
Ok, let's try the following. Please revert to the initial constructor (which produces the mirrored output), then locate u8x8_d_st75256.c on your local driver (arduino lib folder)
and change the following line:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L1096
from
`U8X8_CA( 0xbc, 0x00 ), `
to
`U8X8_CA( 0xbc, 0x02 ), `

Will this help?

GavinBrownAus, 2019-06-22:
Hi Oliver - thats done it. Display is now perfect. Thanks

olikraus, 2019-06-22:
Can you still tell me the name or manufacturer of the display, so that I can create a new constructor?

GavinBrownAus, 2019-06-23:
Hi, details are:

Jinglianxun Electronics JLX256160G-630-BN

Does this help?

olikraus, 2019-06-23:
Yes, thanks.

olikraus, 2019-06-23:
hmmm... I think there are several versions for the JLX256160:
#561 #717 #731

olikraus, 2019-06-23:
I have created beta 2.26.13 with should include the constructor 
`U8G2_ST75256_JLX256160M_F_4W_SW_SPI`

It just includes the above mentioned modification. Maybe you can check this version once. Thanks.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

GavinBrownAus, 2019-06-23:
Hi, getting there. Display is shifted to what looks like about 20
characters to the right. See pic below.

olikraus, 2019-06-23:
There is no picture attached :(

GavinBrownAus, 2019-06-23:
Just sent it direct to you.

olikraus, 2019-06-23:
Hmm 16 bit mode is disabled, I think.

AnHardt, 2019-06-23:
For attachments use githubs web frontend. Link is in every mail.
Attachments do not work via mail.

GavinBrownAus, 2019-06-24:
Spot on 16 bit mode was disabled. All working perfectly now.

Thanks.

olikraus, 2019-06-24:
Great... feel free to reopen if required....

closed 2019-06-24 by olikraus

GavinBrownAus, 2019-09-02:
Hi Oliver,

Quick one: is it possible to speed up the refresh of the U8G2_ST75256_JLX256160M display that you helped with previously. 

The constructor is: U8G2_ST75256_JLX256160_F_4W_SW_SPI

Is there anywhere that I can increase the SPI  speed?

Im using a Teensy 3.5

Any help is appreciated.

olikraus, 2019-09-02:
It all depends on the Arduino API implementation for the Teensy. If the clock speed settings are accepted by the Teensy Arduino HAL, then you can use https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

More precise, the Teensy HAL has to support the SPISettings command: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L773

But at least from the docs, this should work: https://www.pjrc.com/teensy/td_libs_SPI.html

olikraus, 2019-09-02:
hmmm... another precondition is this: Teensy Arduino Environment has to report itself as beeing higher version than 1.6 (10600) : https://github.com/olikraus/u8g2/blob/c2a06c08dca7a70d5834480f1dcfb7ba3664057e/cppsrc/U8x8lib.cpp#L772