olikraus/u8g2 #932: Display routines don't work in a BTCallback: chars on screen scattered and wrong

wcheswick, 2019-06-25:
displayEntry works fine in a normal sketch but not in a BTCallback.  I am betting the environment is different (restricted?) on callbacks. Relevant code fragments:

#define HEADER  (-1)
void displayEntry(int e, char *fmt, ...) {
  assert(MAX_ENTRIES > 0);
  assert(entryColumns > 0 && entryColumns < 6);
  va_list ap;
  char buf[100];

  va_start(ap, fmt);
  vsnprintf(buf, sizeof(buf), fmt, ap);
  va_end(ap);

  int screen = 0;
  if (e == HEADER) {
    u8g2.drawStr(0, 0, buf);
  } else {
    screen = e / MAX_ENTRIES;
    e = e % MAX_ENTRIES;
    int c = e / ENTRIES_IN_COL;
    int r = e % ENTRIES_IN_COL + 1; // top row is the header
    int x = c*(ENTRY_W + 1)*u8g2.getMaxCharWidth();
    int y = r*u8g2.getMaxCharHeight();
    assert(x >= 0 && x < u8g2.getDisplayWidth());
    assert(x + strlen(buf)*u8g2.getMaxCharWidth() < u8g2.getDisplayWidth());
    assert(y >= 0 && y < u8g2.getDisplayHeight());
    Serial.print("@");
    Serial.print(x);
    Serial.print(",");
    Serial.print(y);
    Serial.print("  '");
    Serial.print(buf);
    Serial.println("'");
    
    u8g2.drawStr(x, y, buf);

  // if we have too many entries, show what screen we are on in the
  // upper right corner

    if (screen) {
      snprintf(buf, sizeof(buf), "%+2d", screen);
      int x = u8g2.getDisplayWidth() - 2*u8g2.getMaxCharWidth();
      u8g2.drawStr(x, 0, buf);
    }
  }
  u8g2.sendBuffer();
}

void displayBTDevice(BLEAdvertisedDevice dev) {
  uint nameLen = ENTRY_W - 4;
  std::string addr = dev.getAddress().toString();
  String saddr = &addr.c_str()[sizeof("00:11:22:")-1];
  displayEntry(BTCount, "%*s%4d",
      -nameLen,
      saddr.c_str(),
      (int)dev.getRSSI());
}

class BTCallback: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice dev) {
    std::string advDump = dumpDevice(dev);
    results << advDump << "\n";
    Serial.println(advDump.c_str());
    // access to the display in this callback is broken.  There is probably a good, logical
    // reason for this.
    displayBTDevice(dev);
    BTCount++;
  }
};

olikraus, 2019-06-26:
Not sure how I can support here. This does not look like an Arduino Environment.

olikraus, 2019-08-13:
hmm no further feedback... closing....

closed 2019-08-13 by olikraus

wcheswick, 2019-08-13:
Sorry, I ran out of cycles.  It was for an ESP32, and I neglected to make a demo sketch.