olikraus/u8g2 #933: Can I use OLED 128x128 through SPI with the Mkr Zero using this library?

ancabilloni, 2019-06-28:
I have a [128x128 waveshare OLED display](https://www.waveshare.com/1.5inch-oled-module.htm). Using this library, I can display just fine with an Arduino Mega with setting:

`U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI OLED1(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8)` 
CLK -> pin 52
DIN -> pin 51

Now I want to use the Mkr Zero to display the same thing, but so far I haven't been able to. The setting for my Mkr Zero is

`U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI OLED1(U8G2_R0, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);`
CLK -> pin 9 (SCK)
DIN -> pin 8 (MOSI)

Is it possible for me to use the OLED as mentioned with this library?

Thanks in advance for any help!

olikraus, 2019-06-28:
It should work also with MKR zero boards. Maybe try with software SPI first.

ancabilloni, 2019-06-28:
Thank you for replying. I tried the software SPI but it still doesn't work. Here is the sample code that I'm trying to run (which worked on Mega) 

```
#include <U8g2lib.h>  // U8g2 Library for Oled https://github.com/olikraus/u8g2
 
#define switch_pin1 4  // Tact Switch for 1st OLED connected to pin 8
 
int counter1=0;  // Score counter for Player 1
 
int center1=47;  // Used to Center value for counter display on OLED 1
   
// OLED 1 init
 
 
/// ***Arduino Mega (Working)***
// CLK 52, DIN 51
//U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI OLED1(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
 
// ***Arduino Mkr zero***
//U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI OLED1(U8G2_R0, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);
U8G2_SSD1327_MIDAS_128X128_1_4W_SW_SPI OLED1(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);

//U8G2_SSD1327_WS_128X128_1_4W_HW_SPI OLED1(U8G2_R0, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);
//U8G2_SSD1327_WS_128X128_1_4W_SW_SPI OLED1(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);
 
 
void setup(void) {
  pinMode(switch_pin1, INPUT_PULLUP);  // using Input_Pullup resistor of the Arduino
 
  OLED1.begin();  //  Start OLED 1
  OLED1.setContrast(200);  //  Brightness setting from 0 to 255
 
  // Choose small font for Player 1 text display
  OLED1.setFont(u8g2_font_fub17_tr);
  OLED1.clearBuffer();  // Clear OLED 1
  OLED1.setCursor(10, 32);
  OLED1.print("PLAYER 1");
  OLED1.drawRFrame(0, 7, 128, 32, 7);  // Draw frame around Player 1 text  
  OLED1.setFont(u8g2_font_fub42_tr);  // Choose bigger font for score display
  OLED1.setCursor(center1, 117);
  OLED1.print(counter1);  // Display counter 1 value
  OLED1.drawRFrame(0, 65, 128, 63, 7);
  OLED1.sendBuffer();  // Send to OLED 1
   
}
 
 
void loop() {
 
// If tact switch 1 is clicked
  if (digitalRead(switch_pin1) == LOW) {
    counter1++;  // increase counter
    switch (counter1) {
      case 0 ... 9:   // if counter < 10
        center1=47;   // used to center score in OLED display
        break;
      case 10 ... 99: // if counter between 10 and 99
        center1=32;
        break;  
      case 100 ... 999:  // if counter > 100
        center1=12;
        break;        
    }
    OLED1.clearBuffer();
    OLED1.setFont(u8g2_font_fub17_tr);
    OLED1.setCursor(10, 32);
    OLED1.print("PLAYER 1");
    OLED1.drawRFrame(0, 7, 128, 32, 7);
    OLED1.setFont(u8g2_font_fub42_tr);
    OLED1.setCursor(center1, 117);
    OLED1.print(counter1);
    OLED1.drawRFrame(0, 65, 128, 63, 7);
    OLED1.sendBuffer();
    delay(10);
  }  
}
```

AnHardt, 2019-06-28:
Marlin uses [u8g-lib ](https://github.com/olikraus/u8glib)not [u8g2-lib](https://github.com/olikraus/u8g2). If the dispay isn't [supported](https://github.com/olikraus/u8glib/wiki/device) by [u8g-lib](https://github.com/olikraus/u8glib) the chance to get it work is near zero.

AnHardt, 2019-06-28:
Sorry. Wrong repository.

olikraus, 2019-06-29:
Hmm... difficult to say. 
First I would try to execute one of the existing u8g2 examples. 
Also ensure, that your display is configured as SPI. 
And: Check your wiring

closed 2019-11-23 by olikraus