olikraus/u8g2 #934: ESP32 Hardware SPI

ClintH, 2019-06-29:
Hi,

I'm using u8g2 2.25.10 with a TTGO ESP32 board in the Arduino 1.8.9 environment.  I'm trying to use hardware SPI to drive a SSD1322 256x64 OLED. I only have one display connected, but my ambition is to drive two.

The text displays when I use software SPI, but not when it's using hardware (the display is just dark). From this basis, I assume the display is configured for SPI properly and that it works. 

Is there anything special required to get hardware SPI working on ESP32?

thanks!

```
#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <U8g2lib.h>

#define SPI_CLOCK 14
#define SPI_DATA 12
#define SPI_CS 15
#define SPI_DC 27
 
// This does not work:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0,SPI_CS, SPI_DC);
// This does work:
// U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, SPI_CLOCK, SPI_DATA, SPI_CS, SPI_DC);

void setup() {
  pinMode(SPI_CS, OUTPUT);
  pinMode(SPI_DATA, OUTPUT);
  pinMode(SPI_RESET, OUTPUT);
  u8g2.begin();
}

void loop() {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

olikraus, 2019-06-29:
It should work, but are you sure that you have used proper hardware spi pins for modi and clock?

olikraus, 2019-06-29:
It should work, but are you sure that you have used proper hardware spi pins for modi and clock?

According to what I saw, clk and data are no t at 12 & 14.

ClintH, 2019-06-29:
I was basing the pins from [this reference](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) (jump to the 'SPI' header), and using the HSPI pins (MOSI 13, MISO 12, CLK 14, CS 15).

This doesn't correspond to the best available pin out diagram I have for the board, mind you:

[image]

ClintH, 2019-06-29:
I've also tried to the VSPI pins (MISO 19, CLK 18, CS 5) and updating the code to no avail:

```
#define SPI_CS 5
#define SPI_DC 27
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, SPI_CS, SPI_DC);
```

The OLED display is set to four-pin SPI mode (via SMD resistors), might it be any help to three-pin SPI mode? I'm new to using displays :)

olikraus, 2019-06-29:
Look s more like 18 and 23.
Anyhow, I am the wrong person to answer such questions: what are the Arduino hw spi pins has to answered by the board manufacturer.

olikraus, 2019-06-29:
4 wire SPI is better than 3wire SPI.
U8g2 just is built on the standard Arduino SPI library. If you figure out which pins are used by Arduino SPI.h the just use the same pins.

ClintH, 2019-06-29:
Thanks @olikraus for your patience.

I still didn't get the HSPI lines to work, but I suspect that's because the TTGO ESP32 T8 uses it for the SD card device. I also misunderstood SPI_D/SPI_Q. When the ESP is the controller, V_SPI_D (pin 23) is MOSI, and V_SPI_Q (pin 19) is MISO.

For anyone else in the future, to use the ESP32 VSPI lines, this is how. I'm using a 3.1" LCDSOC M31S1332 256x64 OLED.

[image]

```
#define SPI_CS 5
#define SPI_DATACOMMAND 27
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, SPI_CS, SPI_DATACOMMAND);
```

closed 2019-06-29 by ClintH

olikraus, 2019-06-29:
So, it works now with her spi?

ClintH, 2019-06-29:
Yes, using the ESP32's V_SPI bus rather than H_SPI.

olikraus, 2019-06-29:
Ah, ok. Maybe the second SPI (2ND_SPI).. is mapped to H SPI.

fuxifex, 2019-10-05:
hi,
i've used your settings:

#define SPI_CLOCK 14
#define SPI_DATA 12
#define SPI_CS 15
#define SPI_DC 27

U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, SPI_CLOCK, SPI_DATA, SPI_CS, SPI_DC);

it works on my esp32 DevKitC but it's really sloooooooooow .. about 2fps only.
shouldn't it be really fast? do you know what could be wrong?
shouldn't reset also be wired?
i already checked my board settings, it's set to maximum (240mhz).

thanks

olikraus, 2019-10-05:
You need to use the HW SPI constructor.

kvoit, 2019-10-14:
> Ah, ok. Maybe the second SPI (2ND_SPI).. is mapped to H SPI.

It does not seem so. I tried it and it did not work. Digging a little bit into the code it seems that U8X8_HAVE_2ND_HW_SPI is not defined, so u8x8_byte_arduino_2nd_hw_spi does not do a lot.

Has anybody had some progress here? I have a PCB where the display is already connected to HSPI.

Out of curiosity? Why go als the way to have different classes for SW and two HW SPIs, instead of just passing  a reference to any SPI object to the class?

olikraus, 2019-10-14:
> Out of curiosity? Why go als the way to have different classes for SW and two HW SPIs, instead of just passing a reference to any SPI object to the class?

Good point. I just didn't thought about this option.

> Has anybody had some progress here? I have a PCB where the display is already connected to HSPI.

The problem seems to be, that the u8g2 detection mechanism doesn't work because the ESP32 Arduino files deviate from the standard Arduino mechanism. I think you could still modify the name of the SPI object within the U8g2 lib.