olikraus/u8g2 #935: Drawing vertical lines doesn't always update all pixels

liamlacey, 2019-07-01:
Hi,

I've found an issue where, if I attempt to draw a vertical line that moves fairly quickly across the screen, on some refreshes of the screen it appears that the pixels on some rows (often rows in succession) aren't being updated - screen 'tearing' may be the best word to describe it.

Testing with both an SSD1306 and an SSD1309 module on an Arduino UNO shows the issue fairly subtly, however when using an Atmel SAM chip the issue is a lot more noticeable. 

Is this to be expected?
If not, where should I look / what should I change to try and fix / reduce this?

This is the Arduino code I am testing with, based on the Hello World example:

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

int lineXPos = 0;

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory

  //draw vertical ine
  u8g2.drawLine (lineXPos, 0, lineXPos, 63);
  
  u8g2.sendBuffer();					// transfer internal memory to the display

  //increment line position
  lineXPos++;
  if (lineXPos > 127)
    lineXPos = 0;

  delay(1);
}
```

Thanks in advance.

olikraus, 2019-07-01:
Short answer: sendBuffer is not in sync with screen refresh of the OLED.

Long answer: OLEDS work like almost all other matrix displays. At a time only one line of LEDs is activated. It usually takes around 10ms to refresh all 128 lines of a SSD1306 OLED (so the screen refresh rate is between 60 to 100 Hz): You can also test this with a digi-cam with a very short exposure time.
Note, that this all happens in the display itself, independent of your Arduino code. 

It also means, that there are two full memory buffers: One in your Arduino board and another buffer in the SSD130x controller. sendBuffer will transfer data from the Arduino to the memory of the display controller from where it will take up to 10ms until the display becomes visible.
All in all, it may happen, that the SSD1306 has started to show the new memory content, in the upper part, but the old content from the lower part, because the line update is faster that the transfer from Arduino to memory. This will be visible as tearing effect. 

The problem is described in section 8.4 of the SSD1306 manual. However I do not know any OLED for which this signal is available. so... there is nothing what we can do...

liamlacey, 2019-07-03:
Good to know - many thanks for the quick response and detailed explanation!

closed 2019-07-03 by liamlacey