olikraus/u8g2 #936: S69508-DB025MB LCD 128*64 with ST7565R controller

1simone0, 2019-07-02:
[image]
Helllo olikraus and thank you for your library! 
I am using this display: S69508-DB025MB LCD 128*64 with the following features:ù1-2. FEATURES
(1) Display Type: STN Blue, Negative, Transmissive, 12O’clock
(2) Driving Method: VDD=3.3V, 1/65 duty, 1/9 bias
(3) Built-in controller: ST7565R
(4) With White Backlight.
I used that with Arduino with using the constructor U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, 52, 51, 53, 3, 2); and adjusting the contrast I've had optimal solution!
Now I should use the same display in a stm32 board using the stm32cube IDE ( and C language), so I follow your guide  "porting to new mcu platform". I wrote my  u8x8_stm32_gpio_and_delay function like this:
`uint8_t u8x8_stm32_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8,
    U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,
    U8X8_UNUSED void *arg_ptr)
{
  switch (msg)
  {
  case U8X8_MSG_GPIO_AND_DELAY_INIT:
    HAL_Delay(10);
    break;
  case U8X8_MSG_DELAY_MILLI:
    HAL_Delay(arg_int);
    break;
  case U8X8_MSG_GPIO_DC:
	  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_9, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET:
	  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_8, arg_int);
    break;
  case U8X8_MSG_GPIO_CS:
	  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_6, arg_int);
    break;
  case U8X8_MSG_GPIO_D0:
  	  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_5, arg_int);
      break;
  case U8X8_MSG_GPIO_D1:
      //HAL_GPIO_WritePin(OLED_DC_GPIO_Port, OLED_DC_Pin, arg_int);
  	  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_7, arg_int);
      break;
  }
  return 1;
}
`

but I see the program go in a loop (or maybe crash) when enter the function  u8g2_InitDisplay(&u8g2); 

So because I had already the project ready for work with taht display (but not with the library) I used those init functions to initialize the display and then I tryed to use the library functions to write something. 
Here are the init functions: 
`	io_lcd_pwm_set(5000);
	lcd_init();
	lcd_clear();
	lcd_set_inverse(0);`
where lcd_init(); is this:
`void lcd_init()
{
	io_lcd_reset(0);
	io_lcd_cs(1);
	io_lcd_sda_ON();
	io_lcd_scl_ON();
	io_lcd_rs(0);
	wait_millisec(10);
	io_lcd_reset(1); // exits from reset...
	io_lcd_rs(0);    // commands

	lcd_write_byte(0xa2); /* 1/9 bias */
	lcd_write_byte(0xe2);
	lcd_write_byte(0xa0); /* ADC select , Normal */
	lcd_write_byte(0xc8); /* Common output reverse */
	lcd_write_byte(0xa6); /* normal display 1=on */
	lcd_write_byte(0x2c);
	lcd_write_byte(0x2e);
	lcd_write_byte(0x2f); /* V/C off, V/R off, V/F on */
	lcd_write_byte(0xf8); /***4 booster***/
	lcd_write_byte(0x00);
	lcd_write_byte(0x24); /* internal resistor ratio */
	lcd_write_byte(0x81); /* electronic volume mode set */
	lcd_write_byte(0x1C); /* electronic volume */
	lcd_write_byte(0x40); /* display start first line */
	lcd_write_byte(0xaf); /* display on */
	lcd_write_byte(0xa5); // all pixels ON
	wait_millisec(1000);
	lcd_write_byte(0xa4); // all pixels OFF
	wait_millisec(1000);

}`
But as result I get the Hello world! and circle in wrong x-y position.
Attached the image...
What do you suggest?

BR,

Simone

olikraus, 2019-07-02:
> What do you suggest?

You should follow the instructions in the porting guide. What is your other code to setup the display with the callback function from above? Can you send a complete example which leads to the loop (as mentioned above)

1simone0, 2019-07-02:
What do you mean with the other code to setup the display with the callback function? That's my callback function, I wrote that after reading this guide [https://elastic-notes.blogspot.com/2018/10/u8g2-library-usage-with-stm32-mcu.html?showComment=1562033373431#c8203534280303964941](url)
I put some brakpoint and discovered that in file u8x8_dislay.c , in function u8x8_d_helperr_display_init(u8x8_t *u8x8)  for sure called from u8g2_InitDisplay(&u8g2); my program crash.
In fact, I put a breakpoint on line 67 and another one on line 68
[image]
but I never reach the breakpoint on line 68.
Hope it can help you to understand....

olikraus, 2019-07-02:
Didn't you use one of the setup functions from here?
https://github.com/olikraus/u8g2/wiki/u8g2setupc#st7565-nhd_c12864

My question is: Which function did you use from that table and what are your  arguments.

1simone0, 2019-07-02:
Oh, yes.  I use  u8g2_Setup_st7565_nhd_c12864_1 on Arduino but I also tried u8g2_Setup_st7565_nhd_c12864_2 and u8g2_Setup_st7565_nhd_c12864_f on stm32 with the same results....
my arguments are:
`u8g2_Setup_st7565_nhd_c12864_f(&u8g2, U8G2_R2, u8x8_byte_4wire_sw_spi, u8x8_stm32_gpio_and_delay);` 
I remember you (to be clear) that with the following code I see what I want to write but in a wrong position:
`int main(void)
{

	InitHW();
	io_lcd_pwm_set(5000);
	lcd_init();
	lcd_clear();
	lcd_set_inverse(0);

	u8g2_Setup_st7565_nhd_c12864_f(&u8g2, U8G2_R2, u8x8_byte_4wire_sw_spi, u8x8_stm32_gpio_and_delay);
	//u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
u8g2_FirstPage(&u8g2);
			do
			    {
			      u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
			      u8g2_DrawStr(&u8g2, 0, 10, "Hello World!");
			      u8g2_DrawCircle(&u8g2, 64, 50, 10, U8G2_DRAW_ALL);
			    } while (u8g2_NextPage(&u8g2));`
and with the following code my program don't reach the line 68 of the previous image:
`InitHW();
        //io_lcd_pwm_set(5000);
	//lcd_init();
	//lcd_clear();
	//lcd_set_inverse(0);

	u8g2_Setup_st7565_nhd_c12864_f(&u8g2, U8G2_R2, u8x8_byte_4wire_sw_spi, u8x8_stm32_gpio_and_delay);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
u8g2_FirstPage(&u8g2);
			do
			    {
			      u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
			      u8g2_DrawStr(&u8g2, 0, 10, "Hello World!");
			      u8g2_DrawCircle(&u8g2, 64, 50, 10, U8G2_DRAW_ALL);
			    } while (u8g2_NextPage(&u8g2));`

olikraus, 2019-07-03:
From my perspective init should be good and sufficient.
```
u8g2_Setup_st7565_nhd_c12864_f(&u8g2, U8G2_R2, u8x8_byte_4wire_sw_spi, u8x8_stm32_gpio_and_delay);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
```

Also have a look here: https://github.com/olikraus/u8g2/blob/master/sys/avr/as7/main.c#L174

No other calls  (lcd_init()) should be required.

1simone0, 2019-07-05:
Yes, I thought the same and I discovered the problem was in the `HAL_Delay(10);` function!!! I'm not sure why but it wasn't working properly.... So that problem is solved. Now I could have one more issue to fix: rendering the display take more or less half second. Is it normal?
This is my code. I don't have any delay around....
`		if(!io_sw4())
		{
			//io_lcd_pwm_set(0);
			u8g2_FirstPage(&u8g2);
			u8g2_SetContrast(&u8g2,170);
			do
			    {
			      u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
			      u8g2_DrawStr(&u8g2, 0, 14, "Hello World!");
			      u8g2_DrawCircle(&u8g2, 64, 50, 10, U8G2_DRAW_ALL);
			    } while (u8g2_NextPage(&u8g2));
		}`
where sw4 is a button.
Any suggestion?

olikraus, 2019-07-05:
In the above example, you could move the SetFont command out of the loop, but this will not help much I guess. Finally your `u8x8_stm32_gpio_and_delay` function will be called and the individual bit values are putput manually with GPIO commands (again, not sure how much time is consumed there).

For better speed you should consider to use the SPI hardware subsystem of your controller, so you need to implement your own HW SPI transmit callback:

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication

I mean, you should replace `u8x8_byte_4wire_sw_spi` by a custom procedure which uses your hardware.

closed 2019-08-13 by olikraus