olikraus/u8g2 #937: SSD1327 issues with display inverting, vertical scrolling, and incorrectly interlaced flipped output?

deezums, 2019-07-02:
Hello!

First, thank you for the great library and all the amazing documentation! 

I have a round 128*128 OLED driven by a SSD1327 - CrystalFontz CFAL128128B0-011W
 
I tried using the "EA W128128 round OLED" definition, but the display is offset vertically a good bit. The "MIDAS MCOT128128C1V-YM" definition seems to work almost perfect!

I am running the display off a teensy 3.6, and it's probably important to mention that this is all mounted in a car, the display is in the instrument cluster and the teensy is 3ft of wire away in the center console. The teensy pulls data over canbus and then displays it on the oled. It looks something like this. 

https://www.youtube.com/watch?v=5pkusOi-U0U&feature=youtu.be

For the most part everything works OK, except occasionally the display will distort and continue running all glitched out. Rarely it will glitch out again and fix itself. The most frequent issue is the display inverting colors, followed by the display inverting while alternating rows of pixels have the wrong horizontal offset. What seems strange to me is I rarely ever see errors in the display, no snow or static. It almost makes me think I could adjust some timings in the library to make it work better if I knew what I was doing! 

If I understand u8x8_ssd1327.c correctly, I have tried 24MHz, 20MHz, 16MHz, and 8MHz and all seem to show the same errors. Here is how I have things set now. 

/*  MIDAS MCOT128128C1V-YM 128x128 Module */

static const u8x8_display_info_t u8x8_ssd1327_128x128_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 10,
  /* reset_pulse_width_ms = */ 100, 	
  /* post_reset_wait_ms = */ 100, 		/**/
  /* sda_setup_time_ns = */ 100,		/* */
  /* sck_pulse_width_ns = */ 63,	/*  */
  /* sck_clock_hz = */ 16000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 1,	/* use 1 instead of 4, because the SSD1327 seems to be very slow */
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 40,	
  /* tile_width = */ 16,
  /* tile_hight = */ 16,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,		
  /* pixel_width = */ 128,
  /* pixel_height = */ 128
};

Are there any other places to play with timings?

 Any and all advise is greatly appreciated! Although I'm thinking I just learned the hard way why cars use canbus for communication over any distance...

olikraus, 2019-07-03:
Thanks for sharing the video. Amazing project.
I think you are allready at the right place for the timing, however the offset of the inversion mode is only transmitted once during startup. In fact I read several user feedbacks, where the display controller seems to forget a part of the init values.

The problem could also be, that the environment makes some bits to change in the config memory of the display controller. Usually the display controller is placed on the display itself (Chip on glass technology).
See here for example:
[image] 

The silver bar in the middle of the black area in the upper part of the display is the controller. This part often is exposed and not very good protected. Maybe a small piece of metal could improve here.

Another solution could be to resend the init values from time to time (for example if you change the graph)
This could be done with u8g2::initDisplay(). But there is one problem: initDisplay() will issue a reset as a first activity, which will usually take a lot of time, but mabe you could at least test this a little bit, whether this helps.

olikraus, 2019-07-03:
If you want to resend the init values only, then I assume I need to modify initDisplay() a little bit.
Actually it does three actions for initDisplay:
1) Init the Microcontroller GPIOs
2) Reset the display
3) Send the display startup values to the display controller

initDisplay() is part of begin() and obviously we want all three actions to be executed during power on reset.
But there are some occations where only some of these three activities are required. I try to add a summary for the behavior of initDisplay() depending on the situation:

Power on reset:
1) Init the Microcontroller GPIOs
2) Reset the display
3) Send the display startup values to the display controller

uC Wakeup from deep sleep (display still was powered):
1) Init the Microcontroller GPIOs
3) Send the display startup values to the display controller

Reactivate display from hard fault:
2) Reset the display
3) Send the display startup values to the display controller

Recalibrate display (due to lost configuration information):
3) Send the display startup values to the display controller

If you want, I can update the u8g2 to add the required addons for initDisplay().
Will this be an option for you?
Again: This initDisplay() will not solve the problem, but may reduce the probability that this happens and will also allow the display to heal from a lost config event.

deezums, 2019-07-03:
Thank you for the reply and compliments! Still wouldn't have been possible without all the amazing work you've put into this library!

As for the chip on glass, I do not think this is the problem though I have heard of it. First, the LCD itself has black tape over the chip from the factory, next I have a 0.6mm PCB with WS2812 over, as well as a stainless trim ring to blend the WS2812's a little bit. I haven't considered that the LEDs might be causing a problem, they and the data line are directly above the chip, could this be a problem?

I am leaning towards noise on data lines, and I figure the reason I hardly ever see errors in the display only in the config is because the actual display is pretty much 4x as dense, and 3x as much data can be corrupted with no real discernible change in the display (greyscale changes) 

If it is not too much work for an obscure issue I would love to try those init changes! I know the display might still glitch out with no other changes, but if I add a re-init between "pages" on the display I can just switch between and fix any issues. I am trying to run 18Hz GPS on the same board, with the intention of using it for a lap timer for track days. I can't have too many interruptions happening, as is I think I have a loop speed fast enough to manage, once I relearn trig...

I do think that moving forward I will split my project into two microcontrollers, one that gathers all data, and another that displays it. In the meantime a software fix would be nice!

What is strange is it seems to be much worse at 4MHz than 8MHz, but after reading last night I think it may have to do with teensy SPI library overriding clock speeds set in U8G2. I think the teensy only does hardware spi in divisions of a clock speed, not necessarily CPU clock however. I might run some tests to see if I am getting the clock frequency I think I'm getting. If this is what's happening, I think I can fix it by setting the timings to an even multiple, so teensy's "nearest less than frequency" is equal? 

Here is a photo of the LCD hardware
https://i.imgur.com/GYYQhDU.jpg

And, mind the smudge, but this is how it looks when it's working right! Seems to be a random fluke, but it ran there for ~45 minutes uninterrupted at "24MHz" in u8g2 config. Lately 10-15 minutes is a lucky break!

https://i.imgur.com/rnwc5P0.jpg

Here is the code and hardware, although slightly behind what I am using now. I will update it later tonight...

https://github.com/deezums/TeensyMS

Thanks again for your replies!

olikraus, 2019-08-09:
closing this... feel free to reopen for further questions...

closed 2019-08-09 by olikraus