olikraus/u8g2 #938: Display "arctifacts"

RafaelArkenos, 2019-07-03:
Hello there! 
I need a little help, I'm trying to figure out what kind of problems is these "artifacts" on the screen:
[image]

Everything starts running very well.. but if i wait 10 min+ in the same screen without exchanging the screen, starts to show these "artifacts"
I'm using the u8g2 in a project using a Winstar display - "WEO128128BWPP3N00000" configured whit these parameters:

- Communication trought SPI;
- Setup function: u8g2_Setup_ssd1327_midas_128x128_f

Thanks in advance!

olikraus, 2019-07-03:
Strange: Same display causes similar problems here: #937 

But I think your problem is still a little bit different. I personally would guess, that there is a power supply problem. Perhaps add some extra caps next to the power supply lines of your OLED. 
Maybe ceramic/multilayer caps 100uF and 1uF in parallel. Both as close as possible to the OLED connector.

RafaelArkenos, 2019-07-05:
> Strange: Same display causes similar problems here: #937
>
> But I think your problem is still a little bit different. I personally would guess, that there is a power supply problem. Perhaps add some [...]

Yo! i followed your steps and using a dedicated 3.3v supply. I let the display in tests for 1 day and I think it's ok now. 
Thank you for your dedication and fast support. Your library is awesome!

closed 2019-07-05 by RafaelArkenos

olikraus, 2019-07-06:
:)