olikraus/u8g2 #939: LC7981 compatible LCD - 128x32 based on HD61830A00

Gun-neR, 2019-07-04:
Hello, I have an LCD from some old medical monitor that I have been able to get working off an UNO in a very limited scale, with lots of manual efforts without a library.  It has a HD61830A00 chipset and a 128x32 display.

But with your library I saw that you have a supposedly compatible chip, LC7981, and I get the "best" result with the U8G2_LC7981_160X80 option (well, not actual text, but what looks like parts of the horizontal layouts flashing across the screen).  I suspect the issue is your lowest available resolution is too much for my display.

How can I adjust that driver for 128x32?

olikraus, 2019-07-04:
> Or are you able to add support for that ?

sure. Maybe we first modify the existing 160x80 driver. Once we have a working version, I create the new driver.

Please locate u8x8_d_lc7981.c on your local system.
Then modify the following values:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L205
change to 16

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L206
change to 4

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L209
change to 128

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L210
change to 4

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L221
change to "U8X8_CA(0x02, 128/8-1), "

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L222
change to
U8X8_CA(0x03, 31),

this last value, I am not sure, it might be also "U8X8_CA(0x03, 32)," or something else. This has to be tested what works best.

olikraus, 2019-07-04:
Update:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L210
must be 32 of course...

Gun-neR, 2019-07-04:
Hello, thank you for your efforts.

I may be leading us both down the wrong path with the LC7981 assumption??  I made the edits above, but was getting no results except a blank screen... even after realising and adjusting for a **120**x32 resolution (I counted this time, ha!  ).

For what it is worth, this is the code I was able to modify to get limited text results... perhaps this might allow you to determine if this is a valid display to adapt to or not?

`
/* initialization sequence and cls function taken from lcd_hd61830_lib, cdragos69@yahoo.com
   http://www.skippari.net/lcd/sekalaista/lcd_hd61830_lib.zip
   found from http://forum.lcdinfo.com/viewtopic.php?t=451

   Edited by Gunner to work on older 120x32 LCD with HD61830A00
*/

// pin assignment - 

const byte RS_PIN = 2;
const byte RW_PIN = 3;
const byte E_PIN = 4;
const byte DB0_PIN = 5;
const byte DB1_PIN = 6;
const byte DB2_PIN = 7;
const byte DB3_PIN = 8;
const byte DB4_PIN = 9;
const byte DB5_PIN = 10;
const byte DB6_PIN = 11;
const byte DB7_PIN = 12;

void setup() {
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);

  // Permanently tied up E pin
  digitalWrite(E_PIN, HIGH);

  delay(2000);
  com(word(0x00, 0x3C)); // Display ON, master mode, cursor on/blink, text mode, internal CG-ROM
  com(word(0x01, 0x75)); // 6x8 font (including spacing)
  com(word(0x02, 0x13)); // 20 horizontal characters
  com(word(0x03, 0x1F)); // 128/4 duty cycle (32 lines)
  com(word(0x04, 0x07)); // cursor on 8th line (of character)
  com(word(0x08, 0x00)); // Display start address low
  com(word(0x09, 0x00)); // Display start address high
  com(word(0x0A, 0x00)); // Cursor address low
  com(word(0x0B, 0x00)); // Cursor address high

  cls();

  com(word(B00001100, B01001000)); // H
  com(word(B00001100, B01100101)); // e
  com(word(B00001100, B01101100)); // l
  com(word(B00001100, B01101100)); // l
  com(word(B00001100, B01101111)); // o
  com(word(B00001100, B00100000)); // space
  //com(word(B00001100, B01110111)); // w
  //com(word(B00001100, B01101111)); // o
  //com(word(B00001100, B01110010)); // r
  //com(word(B00001100, B01101100)); // l
  //com(word(B00001100, B01100100)); // d
  com(word(B00001100, B01000111)); // G
  com(word(B00001100, B01110101)); // u
  com(word(B00001100, B01101110)); // n
  com(word(B00001100, B01101110)); // n
  com(word(B00001100, B01100101)); // e
  com(word(B00001100, B01110010)); // r
  com(word(B00001100, B00101110)); // .
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B01001000)); // H
  com(word(B00001100, B01101111)); // o
  com(word(B00001100, B01110111)); // w
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B01100001)); // a
  com(word(B00001100, B01110010)); // r
  com(word(B00001100, B01100101)); // e
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B01111001)); // y
  com(word(B00001100, B01101111)); // o
  com(word(B00001100, B01110101)); // u
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B01100100)); // d
  com(word(B00001100, B01101111)); // o
  com(word(B00001100, B01101001)); // i
  com(word(B00001100, B01101110)); // n
  com(word(B00001100, B01100111)); // g
  com(word(B00001100, B00100000)); // space
  com(word(B00001100, B01110100)); // t
  com(word(B00001100, B01101111)); // o
  com(word(B00001100, B01100100)); // d
  com(word(B00001100, B01100001)); // a
  com(word(B00001100, B01111001)); // y
  com(word(B00001100, B00111111)); // ?
}

void loop() {
}

void cls() {
  // clear screen
  com(word(0x0A, 0x00)); // Cursor address low
  com(word(0x0B, 0x00)); // Cursor address high
  for (int i = 0; i < 80; i++) com(word(0x0C,0)); // reset all text memory
  com(word(0x0A, 0x00)); // Cursor address low
  com(word(0x0B, 0x00)); // Cursor address high
}

void com(word dat) {
  digitalWrite(13, HIGH); // data flow indication on
  digitalWrite(RW_PIN, LOW);
  digitalWrite(RS_PIN, HIGH);
  digitalWrite(DB7_PIN, bitRead(dat, 15));
  digitalWrite(DB6_PIN, bitRead(dat, 14));
  digitalWrite(DB5_PIN, bitRead(dat, 13));
  digitalWrite(DB4_PIN, bitRead(dat, 12));
  digitalWrite(DB3_PIN, bitRead(dat, 11));
  digitalWrite(DB2_PIN, bitRead(dat, 10));
  digitalWrite(DB1_PIN, bitRead(dat, 9));
  digitalWrite(DB0_PIN, bitRead(dat, 8));
  digitalWrite(RW_PIN, LOW);
  digitalWrite(RS_PIN, LOW);
  digitalWrite(DB7_PIN, bitRead(dat, 7));
  digitalWrite(DB6_PIN, bitRead(dat, 6));
  digitalWrite(DB5_PIN, bitRead(dat, 5));
  digitalWrite(DB4_PIN, bitRead(dat, 4));
  digitalWrite(DB3_PIN, bitRead(dat, 3));
  digitalWrite(DB2_PIN, bitRead(dat, 2));
  digitalWrite(DB1_PIN, bitRead(dat, 1));
  digitalWrite(DB0_PIN, bitRead(dat, 0));

  digitalWrite(13, LOW); // data flow indication off
  //wait for busy flag to go away
  digitalWrite(E_PIN, HIGH);
  digitalWrite(RW_PIN, HIGH);
  digitalWrite(RS_PIN, HIGH);
  do {
    delay(10);
  } while (digitalRead(DB7_PIN) == HIGH);
}
`

Gun-neR, 2019-07-05:
Well, I have been poking various numbers into various places, sometimes randomly, other times with a modicum of intelligence about actual HEX/Binary/decimal conversions ;)

But I have not seen anything more than variations of flashing "static".  I am probably way off on my refresh timing, memory locations for graphics mode, or some other issue.

This is why I use libraries, not make them :P   I think I must await input from someone smarter...

olikraus, 2019-07-05:
Maybe also the timing needs to be adjusted. I see that your code checks the busy flag. 

> This is why I use libraries, not make them :P I think I must await input from someone smarter...

Yes, true, it is difficult for me not having the display in my lab.

Gun-neR, 2019-07-05:
Hmm, yes... I am still trying to figure out how all this bit banging, or whatever it is doing, works... and I am still befuzzled on how to kick this thing into graphics mode... 

> I see that your code checks the busy flag.
I found that if I reduce that wait to even 1ms everything still works, but much faster on the redraw... doh...  However, removing even that little 1ms delay scrambles everything and leaves random characters all over the screen.  I wonder if your library is just sending too much too fast for this old display?

olikraus, 2019-07-05:
> I wonder if your library is just sending too much too fast for this old display?

Yes, this is exactly what I mean.
You could add extra delays here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L196
and here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L197
and here for HW SPI:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L198

Note that the values must not exceed 255.

olikraus, 2019-07-05:
Oh, excause, your display has a parallel interface. Then it is here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L203
and here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c#L204

Gun-neR, 2019-07-06:
> it is difficult for me not having the display in my lab.

Yes, I can understand... it is almost equally difficult for me with the device, but no solid background in control... the experience for which is the only reason I am even trying with this ancient thng.

> You could add extra delays

None of the timing changes made any visible differences, so I guess there is still somthing more missing with this older HD61830A chipset, compared to the supposedly compatible LC7981.

Until my frustration exceeds my interest, I will continue to poke around with it and let you know if I stumble across somthing that can assist in a proper driver development for u8g2

olikraus, 2019-10-03:
Not sure how to continue here... Closing this for now... can be reopened at any time...

closed 2019-10-03 by olikraus

Gun-neR, 2019-10-05:
Oh, sorry... I wasn't getting any further in my efforts so I had gave up... and then forgot all about it :) I should have closed this sooner.

olikraus, 2019-10-05:
No problem...