olikraus/u8g2 #940: Frage/Question Font

Robiv8, 2019-07-04:
Hello,
I have an older sketch that runs with u8g2 but contains older fonts.
Is there a way to find out which of the u8g2 fonts fit.
I could not find anything yet. That would
u8g_font_5x8
u8g_font_4x6
u8g_font_tpss
u8g_font_6x13B

Many Thanks

Hallo,
Ich habe ein älteren Sketch der zwar mit u8g2 läuft aber noch ältere Fonts beinhaltet.
Gibt es eine Möglichkeit rauszufinden welche der u8g2 Fonts dazu passen.
Ich konnte bisher nichts finden. Das wäre
u8g_font_5x8
u8g_font_4x6
u8g_font_tpss
u8g_font_6x13B

Vielen Dank

olikraus, 2019-07-05:
Es gibt eine mapping table:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L2762
Bis auf den tpss müsste es alle fonts auch in u8g2 geben.

den tpss müsste ich eigentlich noch generieren können (er ist nicht in u8g2 enthalten)
Die die sourcen waren hier: 
https://github.com/olikraus/u8glib/blob/master/tools/font/build/do_u8g_fonts.sh#L362
bzw
https://github.com/olikraus/u8glib/blob/master/tools/font/ttf/TEACPSS_.TTF

mal gucken wann ich dazu zeit finde...

Robiv8, 2019-07-05:
WoW Klasse, hilft schon mal weiter!
Dankeschön

olikraus, 2019-07-06:
Anbei der font. Einfach in dein .ino file hineinkopieren. Achtung: Der name ist etwas anders.
Dafür belegt das neue u8g2 font format 500 bytes weniger in deinem Speicher.

```
const uint8_t u8g2_font_tpss_tn[2005] U8G2_FONT_SECTION("u8g2_font_tpss_tn") = 
  "\263\0\3\2\4\4\2\5\5\13\21\377\374\11\374\11\0\1X\2\257\7\270 \5\0B\12!\7\221\303"
  "\211\213\0\42\7#\337\212D\11#\13U\303\233\312\240T\6\245\2$\20\225\303\233\245\242d[\22%"
  "J\242T\26\0%\15\225\303\213L\12\263\260\26FZ\0&\21\227\303\234\255\224e[)\213\262(K"
  "\222%\1'\6!\337\11\1(\14\225\303;)\13\263\64LS\1)\14\225\303\13\265\230fa\26i"
  "\0*\13u\307\253Je\313\226\246\10+\12U\307\253\60\32\244\60\2,\6\42\77\32\5-\6\25\317"
  "\213A.\6\21\303\211\0/\14\225\303\313,\314\302Z\230\205\0\60\12\225B\233%\363[\262\0\61\13"
  "\225B\253LJ\302\236\6\1\62\13\225B\233%\323\302\254\343 \63\15\225B\233%\13#\65\14\265d"
  "\1\64\11\225BKM\33\306\36\65\14\225B\213c\70\210E-Y\0\66\15\225B\233%\23\207$s"
  "K\26\0\67\12\225B\213A\314\32\273\1\70\15\225B\233%\323\222%sK\26\0\71\15\225B\233%"
  "\263%C\30j\311\2:\6A\303\211(;\7R\77\232T\1<\10\225\303\313\254k\7=\10\65\313"
  "\213A\35\4>\11\225\303\213\264[G\0\77\15\225\303\233%\13\263b\16\205\21\0@\25\211\303\235a"
  "\212\223HJJISRR\242\34H\206!\1A\13\225B\233%\323\206!\363\26B\15\225B\213!"
  "\311\264A\311\334\6\5C\13\225B\233%\23{K\26\0D\13\225B\213!\311\374\66(\0E\13\225"
  "B\213c\70$a\343 F\13\225B\213c\70$aG\0G\15\225B\233%\23\223!sK\26\0"
  "H\13\225B\213\314\66\14\231\267\0I\6\221B\211\7J\11\225B\313>j\311\2K\16\225B\213\314"
  "\224\224\264$\252dZ\0L\11\225B\213\260\37\7\1M\14\227B\214t[*R\244z\15N\13\225"
  "B\213lR\22i\363\26O\12\225B\233%\363[\262\0P\14\225B\213!\311\264A\11;\2Q\15"
  "\225B\233%\363\222(\211RY\0R\14\225B\213!\311\264A\311\274\5S\14\225B\233%S\327P"
  "\323\222\5T\11\225B\213A\12\373\11U\11\225B\213\314\337\222\5V\17\225B\213\314\226\224\222(\211"
  "\222,\214\0W\21\227B\214\324\232\64%\245\244\226\204I\230D\0X\15\225B\213LKja\226\324"
  "l\1Y\13\225B\213LKjaO\0Z\13\225B\213A\254u\14\7\1[\11\223\303\212!\352\323"
  "\0\134\12u\303\213\64L\213i\0]\11\223\303\212\251OC\0^\6#\353\232\6_\6\26C\214\3"
  "`\6\42k\32\5a\12eB\233%sK\206\0b\14\225B\213\260\70$\231[\262\0c\12eB"
  "\233%\23k\311\2d\13\225B\313\226AsK\206\0e\14eB\233%\33\206\60K\26\0f\13\225"
  "B+\251\222ma\67\0g\15\245\62\233%sK\206\60\324\222\5h\12\225B\213\260\305\244\271\5i"
  "\7\221B\11e\30j\12\322\262\231Z\322/\12\0k\15\225B\213\260))iIT\311\2l\6\221"
  "B\211\7m\16gB\214D\31\22S$ER$\25n\11eB\213\304\244\271\5o\12eB\233%"
  "sK\26\0p\13\245\62\233%s\33\224\260\21q\22\246\262\233!\311\222,\311\222,\32\322pL\23"
  "\0r\12d\302\212d\210\262\66\0s\12eB\233%]\265d\1t\13\224\302\232\254\64$Y'\0"
  "u\11eB\213\314[\262\0v\14eB\213LKJI\26F\0w\16gB\214TM\232\222Z\22"
  "&\21\0x\13eB\213,\251\205YR\13y\15\245\62\213\314[\62\204\241\226,\0z\11eB\233"
  "A\353\66\10{\14\226CL-L\63\71\215c\1|\6\221\303\211\7}\14\226C\14\271\32ki\230"
  "\211\0~\7%j\213E\31\241\7\221B\11e\30\242\15u\303\253l\251(QR\331\42\0\243\14\225"
  "\303\213\60\34\222\260M\33\6\244\13U\307\213,\231\222h\311\2\245\17\225\303\213LKja\64H\321"
  " E\0\246\6\221\303\211m\247\21\246C\234!\211\207$L\206\70M\305dH\0\250\6\25\317\213,"
  "\251\17w\303\234AI\225E\311\224EM\6\5\252\17\206C\234)K\262$K\262h\211\207\1\253\7"
  "\65\313\233Hj\254\7%\313\213A\14\255\5\0B\14\256\17w\303\234AI\225E\311\224LM\6\5"
  "\257\6\25Z\213A\260\7\63\313\232.\0\261\14u\303\253\60\32\244\60\7\6\1\265\5\0B\14\267\5"
  "\0B\14\273\11\65\313\213\250$%\0\300\16\305B\233\64G\226L\33\206\314[\0\301\15\305B\273:"
  "\262d\332\60d\336\2\302\16\305B\253,\311\201%\323\206!\363\26\303\16\305B\213E\231\227L\33\206"
  "\314[\0\304\15\265B\233:\260d\332\60d\336\2\305\17\325B\253,\311rd\311\264a\310\274\5\306"
  "\23\231B\235\341\26f\341\60(Y\230\205Y\230\205\331 \307\14\225:\233%\23\333\222-\223\0\310\16"
  "\305B\233\64\7\216\341\220\204\215\203\0\311\15\305B\273:p\14\207$l\34\4\312\16\305B\253,\211"
  "\207\61\34\222\260q\20\313\15\265B\233\362\60\206C\22\66\16\2\314\11\302A\211(K\372\3\315\11\302"
  "B\31-\351/\0\316\11\303A\231\306\250\277\0\317\12\263A\211$\214\372\13\0\320\15\225B\213!\311"
  "\264!\321\334\6\5\321\16\305B\213EY\63\333\244$\322f\13\322\14\305B\233\64G\226\314o\311\2"
  "\323\14\305B\273:\262d~K\26\0\324\15\305B\253,\311\201%\363[\262\0\325\14\305B\213E\231"
  "\227\314o\311\2\326\14\265B\233:\260d~K\26\0\327\7\63J\213\244\3\330\22\227B\254\245\26E"
  "RRJJII\252%\311\4\331\14\305B\233\64\7\62\177K\26\0\332\13\305B\273:\220\371[\262"
  "\0\333\14\305B\253,\211\63\177K\26\0\334\12\265B\233r\346o\311\2\335\14\305B\273:\220iI"
  "-\354\11\336\15\225B\213\60\34\222lP\302F\0\337\16\265:\213!\311\24%sQ\302F\0\340\14"
  "\225B\233\64G\226\314-\31\2\341\13\225B\273:\262dn\311\20\342\15\225B\253,\311\201%sK"
  "\206\0\343\14\225B\213E\231\227\314-\31\2\344\13\205B\233:\260dn\311\20\345\15\245B\253,\311"
  "rd\311\334\222!\346\16iB\235K\226i\303\26f\225\13\0\347\13\205:\233%\23k\311V\3\350"
  "\14eB\233%\33\206\60K\26\0\351\14eB\233%\33\206\60K\26\0\352\14eB\233%\33\206\60"
  "K\26\0\353\14eB\233%\33\206\60K\26\0\354\10\242\302\211(Mz\355\10\242\302\31\65\351\5\356"
  "\11\243B\232v \352\5\357\12\223B\212$\7\242^\0\360\14\225B\313hL\6\315-\31\2\361\13"
  "\225B\213EY\23\223\346\26\362\14\225B\233\64G\226\314-Y\0\363\13\225B\273:\262dn\311\2"
  "\364\15\225B\253,\311\201%sK\26\0\365\14\225B\213E\231\227\314-Y\0\366\13\205B\233:\260"
  "dn\311\2\367\12UF\253\34\30t \2\370\14eB\233%Z\22%\231\222\5\371\13\225B\233\64"
  "\7\62o\311\2\372\13\225B\273:\220yK\26\0\373\13\225B\253,\211\63o\311\2\374\12\205B\233"
  "r\346-Y\0\375\16\325\62\273:\220yK\206\60\324\222\5\376\14e\302\212pH\262A\11C\0\377"
  "\16\305\62\233r\346-\31\302PK\26\0\0\0\0\4\377\377\0";
```

olikraus, 2019-07-06:
Hier noch das Übersichtsbild:

[image]

Robiv8, 2019-07-06:
Wahnsinn! Ich danke dir Vielmals.
Top Job
Danke Danke Danke

closed 2019-07-06 by Robiv8