olikraus/u8g2 #941: Inverting Sharp 144x168, 1.3inch LCD

SALLDHMF, 2019-07-05:
You mentioned (Sharp Memory LS027B7DH01 Support # 675) about the color inversion (black/white) for the LS027B7DH01 display - "It does not make sense for monochrome displays to associate 1 with white and 0 with black. The same display can be produced in a positive or negative variant and U8g2 will not know whether you have a positive or negative display."

I have the same problem with the Waveshare 1.3inch memory LCD ls013b7dh05_144x168 display. Using the u8g2 library, text and graphics are white and the background is black. Using the Adafruit_SharpMem.h and Adafruit_GFX.h libraries, text and graphics are black on a white background.

Is it impossible to obtain black text and graphics on a white background using the u8g2 library?

I would like to have some of this and some of it ;-) - Opportunities of the u8g2 library but black on a white background.

Tomasz

olikraus, 2019-07-06:
You could just draw a black  box on the complete screen and then use draw color 0 for text.

SALLDHMF, 2019-07-06:
OKAY. This is a simple way ;-). 
Thanks for the answer.

Tomasz

P.S.
I have to ask for something else, so there will be another issue.

closed 2019-08-13 by olikraus