olikraus/u8g2 #942: Heltec LoRa WAN versus u8g2 :-(

SALLDHMF, 2019-07-06:
I am starting the LoRa connection between two Heltec WiFi LoRa 32 (V2) modules with ESP32 processors - https://heltec.org/project/wifi-lora-32/.

The modules have added OLED displays which are easily operated by u8g2 libraries - U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2 (U8G2_R0, 15, 4, 16); or U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2 (U8G2_R0, 15,  4, 16);

Unfortunately, after the LoR's initialization, neither in full buffer nor page buffer, nothing appears on the display.

Unfortunately, my programming knowledge is small and I have no idea where to look for the cause.

https://github.com/HelTecAutomation/Heltec_ESP32.
https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/LoRa/LoRaReceiver/LoRaReceiver.ino

olikraus, 2019-07-06:
Use u8g2 HW i2c  or use different pins. See FAQ.

SALLDHMF, 2019-07-06:
Hmm ...
Using other pins will not be so easy, all connections are on PCB.
Changing SW to HW causes that there is nothing on the display, also without initiating LoR.

Do I have to make any changes in the code when I change SW to HW?

[WIFI_LoRa_32_V2.pdf](https://github.com/olikraus/u8g2/files/3364679/WIFI_LoRa_32_V2.pdf)

olikraus, 2019-07-06:
Yes, you have to remove the first two pin numbers and keep the last.

SALLDHMF, 2019-07-06:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 16);
And there is nothing on the display, also without initiating LoR.

olikraus, 2019-07-06:
Hmmm, u8g2 just uses the Arduino Wire.h

Do the u8g2 examples work?

SALLDHMF, 2019-07-06:
Yes. With SW and without LoRa i can run all examples on this OLED. 
With ....HW_I2C u8g2(U8G2_R0, 16); there is nothing on the display with or without initiating LoR.

SALLDHMF, 2019-07-06:
I took a step forward.
It was enough to change

Heltec.begin (true / * DisplayEnable Enable * /, true /*Heltec.LoRa Disable * /, true / * Serial Enable * /, true / * PABOOST Enable * /, BAND / * long BAND * /);

on Heltec.begin (false / * DisplayEnable Enable * /, true /*Heltec.LoRa Disable * /, true / * Serial Enable * /, true / * PABOOST Enable * /, BAND / * long BAND * /);

Today, the use of computers is a never-ending series of trial and error.

olikraus, 2019-07-06:
This never had been different :(

SALLDHMF, 2019-07-06:
Yes and no. I professionally programmed 30 years ago and then there was documentation. Today, not only nobody has time to write, but no one has time to read. Only books and movies for beginners are left. Thank you for all the suggestions and for your library. I hope that everything will work now.

olikraus, 2019-07-06:
:)

SALLDHMF, 2019-07-07:
I have a number of trials behind me. Library U8G2_SSD1306_128X64_NONAME_F_SW_I2C works, but it's too slow for me. Refreshing the screen with one (small font) 3-digit number takes about 200 ms.

I should try HW instead of SW - U8G2_SSD1306_128X64_NONAME_F_HW_I2C. Unfortunately, the display is empty in this configuration.

You mentioned that u8g2 uses the Arduino Wire.h. Meanwhile, Wire.h probably assumes that SDA = 21 and SCL = 22. In the case of my module, the display is connected differently - SDA = 4, SCL = 15, and RST = 16.

Is it possible to change the settings of pins in your library?

olikraus, 2019-07-07:
> Is it possible to change the settings of pins in your library?

U8g2 supports pin remapping if this is also supported by Wire.h. So the question is more, is it supported by Wire.h, then (maybe) it is also supported by U8g2. The syntax of the constructor is this:

`U8G2_SSD1306_128X64_NONAME_F_HW_I2C(rotation, reset , clock, data)`

clk and data pin numbers have to be mentioned after the reset pin name (source: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname)

Pin remapping is a none-Arduino extension for ESP controllers. Implementaton still might be incomplete and is also not fully supported by u8g2.

SALLDHMF, 2019-07-07:
1) Is the constructor
U8G2_SSD1306_128X64_NONAME_F_HW_I2C (rotation, reset, clock, date)? 
In my case it will be: U8G2_SSD1306_128X64_NONAME_F_HW_I2C (U8G2_R0, 16, 15, 4);. However, I have an error - expected ')' before ',' token. The compiler expects); after the first parameter.

2) table: "https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname" does not contain constructors for I2C.

Tomasz

olikraus, 2019-07-07:
At least you have to provide the name of the object (let me call it "u8g2"):

`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 16, 15, 4);`

> table: "https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname" does not contain constructors for I2C.

It is a little bit confusing, i agree, but there is an extra table a little bit below.

SALLDHMF, 2019-07-07:
Oops. In fact, in the first case and in the second, it is my inattention.

Yes! Yes! Yes! It works!
The time of executing the u8g2.sendBuffer () command: SW - 174 ms HW - 38 ms.

I am grateful for your help. :-)

You wrote: _"U8g2 supports pin remapping if this is also supported by Wire.h. So the question is more, is it supported by Wire.h, then (maybe) it is also supported by U8g2."_. It is now known that the second hardware I2C can be operated in ESP 32.

Tomasz

olikraus, 2019-07-07:
:)

olikraus, 2019-08-13:
hmmm.. I think we can close this issue...

closed 2019-08-13 by olikraus

StefanBertels, 2019-11-26:
I had a related issue (empty screen) when using u8g2. SSD1306_Wire and Heltec code worked and I could not find a relevant difference in code. After trying any possibly correct constructor and double-checking every pin and parameter I found that adding a `delay` before calling `u8g2.begin()` solved the issue for me.

ElGideon, 2020-08-25:
I have the Heltec ESP32 LoRA WiFi Kit V2 868-915 version and I can't get the u8g2 lib example in this project to work using the constructor U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 16, 15, 4); mentioned here, I also tried other constructors with no success. 
Could it me a problem with the library installation or the ESP32 module's OLED I2C address is different?

dougparkarosa, 2021-11-14:
Thanks for this thread. I needed to add delay(100) before begin(). Not sure what an appropriate delay is.

    U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0, OLED_RESET, OLED_SCL, OLED_SDA);
    delay(100);
    display.begin();

dougparkarosa, 2021-11-14:
While I can get it to work using my previous code, I am getting the following message.

    [E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0

Adjusting the delay didn't help. 
I finally dug through the Heltec code and noted that they were doing something like:

    pinMode(OLED_RESET, OUTPUT);
    digitalWrite(OLED_RESET, LOW);
    delay(50);
    digitalWrite(OLED_RESET, HIGH);

Before calling `Wire.begin(...);`

I added the above before the U8G2...HW... constructor and now it seems to consistently work. I tried it on two different Heltec WIFI LORA V2 boards. I noticed that in U8G libraries that a reset is being done, but after calling Wire.begin(...) (u8x8_dsplay.c u8x8_d_helper_display_init)