olikraus/u8g2 #943: U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C isn’t a constructor?

FGasper, 2019-07-06:
Hello,

Maybe I’m misunderstanding something in the documentation, but when I replace:
```
U8X8_SSD1306_128X64_NONAME_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```
… with:
```
U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C u8x8();
```
… and try to verify the code, Arduino IDE gives me:
> request for member 'begin' in 'u8x8', which is of non-class type 'U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C()'

Here’s my full sketch; is there something I’m missing?
```
#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>

/* Constructor */
//U8X8_SSD1306_128X64_NONAME_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C u8x8(void);

/* u8x8.begin() is required and will sent the setup/init sequence to the display */
void setup(void)
{
  u8x8.begin();
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,0,"Hello World!");
  delay(1000);
}
```
Thank you!

FGasper, 2019-07-06:
It appears that specifying the reset pin is needed. I’m going to leave this issue open in case an update to the docs seems warranted.

Thank you!

olikraus, 2019-08-13:
Hmm.. you skip the reset pin from my perspective, the problem seems to be the "void" argument, which is probably not valid:
`U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C u8x8(void);`
This should be
`U8X8_SSD1305_128X32_NONAME_2ND_HW_I2C u8x8();`
instead.

Actually this is not a u8g2 issue more a general C++ topic.

closed 2019-10-03 by olikraus

hansmbakker, 2023-07-23:
I see this is closed, but I do have the same error when doing
```cpp
U8X8_SSD1306_128X32_UNIVISION_HW_I2C u8x8();
```

olikraus, 2023-07-24:
Which error exactly you have?

hansmbakker, 2023-07-24:
The relevant parts of my code are:
```cpp
#include <U8x8lib.h>
U8X8_SSD1306_128X32_UNIVISION_HW_I2C display();

void printOnDisplay(std::string text, bool keepOnDisplay)
{
  display.drawString(0, 0, text.c_str());
  if (!keepOnDisplay)
  {
    delay(1000);
    display.clearDisplay();
  }
}

void setup()
{
  display.begin();
  printOnDisplay(std::string("Hello"), true);
  delay(3000);
  display.clearDisplay();
}

void loop()
{
}
```

The relevant errors I get are:
[image]

olikraus, 2023-07-24:
Hm strange.
Maybe try:

``` C++
U8X8_SSD1306_128X32_UNIVISION_HW_I2C display(U8X8_PIN_NONE);
```

hansmbakker, 2023-08-11:
Sorry for my late response!
```c
U8X8_SSD1306_128X32_UNIVISION_HW_I2C display(U8X8_PIN_NONE);
```
works!

Which is odd though, because the constructor does have a default parameter value:
https://github.com/olikraus/u8g2/blob/a543c481319f381e91c0caea664b7be5ade59635/cppsrc/U8x8lib.h#L2143

I don't have the C++ experience to understand why this still happens, so unfortunately I cannot create a PR for this.

Would it be possible to update the docs at https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#ssd1306-128x32_univision-1? There the parameter is marked optional.

olikraus, 2023-08-13:
I agree, your constructor should also work. Maybe Arduino IDE changes something here. It runs a preprocessor on the code. :-/