olikraus/u8g2 #944: clock and data on this sh1106 just like we can with ssd1305

ThatBigPrint, 2019-07-08:
hi i ideally need there to be a way to specify the clock and data pins on this sh1106 just like we can with ssd1305 we are using the ssd1306 setting for testing but its obviously shifted on the panel 

like this example of what we want please

`U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);`

also the old appears to be  slightly dim will this be code or hardware ?

kind regards chris

olikraus, 2019-07-08:
The specification of clock and data pins has to be done on the microcontroller side, not on the oled side.
moreover there is not difference between ssd1306 and sh1106 with respect to the constructor variants and arguments.

ThatBigPrint, 2019-07-08:
so in example "a" everything is working aside from the shift to the side but when i use example "b" i get the following error 

I'm using an stm32f103

"no matching function for call to 'U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI::U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI(const u8g2_cb_t*, <anonymous enum>, <anonymous enum>, <anonymous enum>, <anonymous enum>, <anonymous enum>)'
"

Example a

"U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
"

Example b

"U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
"

olikraus, 2019-07-08:
Use SW_SPI for example b

ThatBigPrint, 2019-07-08:
Thank you :) as for brightness is that easy to adjust ? through the code ?

closed 2019-07-09 by ThatBigPrint

olikraus, 2019-07-09:
> as for brightness is that easy to adjust ? through the code ?

Yes: Use the set contrast function.

> Use SW_SPI for example b

Actually you could also use HW_SPI, but most uC (including stm32f103) do not allow changing the pin numbers for hardware driven SPI.