olikraus/u8g2 #945: Slow SSD1327 128x128 1.5" I2C on Adafruit Grand Central

Pedrinbeep, 2019-07-11:
Hi =)

I´m testing this oled with the Adafruit Grand Central M4. It have 256Kb of RAM, enough to go really fast with the constructor _F_, but it super slow. With _1_ or _2_ goes really fast but only use less than 1/4 oh the screen. 
Also I set "u8g2.setBusClock(1000000)" and "u8g2.setBusClock(4500000)". 
I'm testing it with Shennong example. What am I doing wrong? 

Constructor:
U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Velocity:
#define SCROLL_DELTA 1
#define SCROLL_DELAY 10

Video with _F_:
https://youtu.be/rSfQZq7xaC8
Video with _1_:
https://youtu.be/UoubFvwakyk

olikraus, 2019-07-11:
You can not simply change the `_F_` constructor to `_1_` or `_2_`, because the sendBuffer command is not valid for the `_1_` or `_2_` constructors. As a result the screen will not be fully visible.

The speed is an issue mostly because of the amount of data, which needs to be transferred. This is four times higher then required because the display is actually a graylevel display and u8g2 has to transmit four bits to set a single pixel.

olikraus, 2019-07-11:
I once did a small review on a similar OLED and send the results to seeedstudio: https://www.seeedstudio.com/Grove-OLED-Display-1-12-V2.html

Pedrinbeep, 2019-07-11:
Okey, I understand. I will continue try other things to improve it but if nothing works, I will look for another OLED display like the one you send me. 
Thank you for your time Olikraus, and for the library!

Pedrinbeep, 2019-07-11:
Okey just for help other users. With Adafruit Grand Central the displays fly if you don't use that example "Shennong". With  _F_ constructor, on the examples "SelectionList" or "UpdateArea", up to "u8g2.setBusClock(4000000)" the display go really fast. It is something that can be done well with the amount of data that needs to be transferred on "Shennong". 

If someone need me to try anything with this board, just say it.

Thank again Olikraus!

olikraus, 2019-08-09:
Indeed, the SSD1327 device is preconfigured with 100KHz for the I2C Bus.
Adding "u8g2.setBusClock(4000000)"  into setup() can speed up the display a lot.

closed 2019-08-09 by olikraus