olikraus/u8g2 #946: openicon Font characters

trkmml, 2019-07-13:
hi i try to use "u8g2_font_open_iconic_all_2x_t" package on nodemcu,
i find https://github.com/olikraus/u8g2/wiki/fntgrpiconic#open_iconic_all_2x this page but
i cant understand 
how to use this icons with this code
u8g2.drawStr( 0, (16 * 1), "????");
with ascii codes or alphabetical character to create icons?
which character corresponds to which icon

[image]

olikraus, 2019-07-14:
You can use the hex escape sequence inside a C string (https://en.wikipedia.org/wiki/Escape_sequences_in_C)

For example key has the decimal code 193, which is the hex code c1. The string would be "\xc1".

If you would use the drawUTF8 function, then instead use "\u" together with the four digit hex number.

trkmml, 2019-07-14:
thank you. 
i can create now with your code
u8g2.drawUTF8( 0, (16 * 1), "\u007b");

closed 2019-07-14 by trkmml

olikraus, 2019-07-14:
:)