olikraus/u8g2 #948: U8g2 when using more than one display type ...

kaspencer, 2019-07-14:
Has anyone any experience of how the u8g2 library behaves when more than one type of display is used with an acceptable Arduino device?
For example:
Several SSD1306 (7-pin) devices on the Hardware SPI bus, selectable by their CS pins each on a distinct pin;
and
One or more SH1106 (7-pin) devices also on the Hardware SPI bus, and also selectable by one or more CS pins each unique to one such device.
If anone has been able to generate constructors for both of these device types that enable each to work without affecting the other, I would be interested to hear of their experiences. 
Best wishes,
Kenneth Spencer

olikraus, 2019-07-15:
Did you look at the FAQ? https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L236

One problem is a shared reset line: Whenever you init a display, all others will receive the reset pulse.

kaspencer, 2019-07-15:
That's a helpful answer Ollie and I will investigate further.
On my project, 20 x SSD1306 OLEDs share the Hardware SPI, the DC and the RESET each with a separate CS pin. They are individually addressed by enabling their CS pins. That works perfectly.
When I introduce the SS1106, which shares the Hardware SPI, DC and RESET pins with the SSD1306s,  the SSD1306's are RESET, so that looks like the answer.
I will separate the RESEST lines for the two groups of devices.
Best wishes,
Kenneth Spencer

olikraus, 2019-07-15:
You can also specify the RESET pin number to one constructor only. Then init (call to begin) this constructor first. As a result all displays receive the reset. After this, call the begin() function of all other displays.

kaspencer, 2019-07-15:
The following section of this message contains a file attachment
prepared for transmission using the Internet MIME message format.
If you are using Pegasus Mail, or any other MIME-compliant system,
you should be able to save it or view it from within your mailer.
If you cannot, please ask your system administrator for assistance.

   ---- File information -----------
     File:  DEFAULT.BMP
     Date:  21 Sep 2001, 14:59
     Size:  358 bytes.
     Type:  Unknown

olikraus, 2019-10-03:
hmm.... closing this...

closed 2019-10-03 by olikraus