olikraus/u8g2 #950: why U8G2_ST75256_JLX25664 256x64 using only 240 pixels in X

mulorc, 2019-07-17:
Hi thanks , this lib is very powerful

I have a JLX25664

this my code:

void setup(void) {  
  u8g2.begin();  
}

void loop(void) {

  draw("What a beautiful day!", SUN, u8g2.getDisplayWidth()); // here getDisplayWidth() = 240

  draw("The sun's come out!", SUN_CLOUD, u8g2.getDisplayHeight());

  draw("It's raining cats and dogs.", RAIN, 8);
  draw("That sounds like thunder.", THUNDER, 12);
  draw("It's stopped raining", CLOUD, 15);
}

I found code in function 

static void u8g2_update_dimension_common(u8g2_t *u8g2)
{
  const u8x8_display_info_t *display_info = u8g2_GetU8x8(u8g2)->display_info;
  u8g2_uint_t t;
  
  t = u8g2->tile_buf_height;
  t *= 8;
  u8g2->pixel_buf_height = t;
  
  t = display_info->tile_width;
#ifndef U8G2_16BIT
  if ( t >= 32 )
    t = 31;
#endif
  t *= 8;
  u8g2->pixel_buf_width = t;
  
  t = u8g2->tile_curr_row;
  t *= 8;
  u8g2->pixel_curr_row = t;
  
  t = u8g2->tile_buf_height;
  /* handle the case, where the buffer is larger than the (remaining) part of the display */
  if ( t + u8g2->tile_curr_row > display_info->tile_height )
    t = display_info->tile_height - u8g2->tile_curr_row;
  t *= 8;
  
  u8g2->buf_y0 = u8g2->pixel_curr_row;   
  u8g2->buf_y1 = u8g2->buf_y0;
  u8g2->buf_y1 += t;

  
#ifdef U8G2_16BIT
  u8g2->width = display_info->pixel_width;
  u8g2->height = display_info->pixel_height;
#else
  u8g2->width = 240;  // <--------------------------------- ????????
  if ( display_info->pixel_width <= 240 )
    u8g2->width = display_info->pixel_width;
  u8g2->height = display_info->pixel_height;
#endif

}

mulorc, 2019-07-17:
[image]

mulorc, 2019-07-17:
Using the u8X8 library, it's perfect

U8X8_ST75256_JLX25664_4W_SW_SPI u8x8(PB9, PB8, PC13, PB6, PB2);

[image]

olikraus, 2019-07-17:
You must activate 16 Bit mode. If U8g2 is not in 16 bit mode, it will limit the width to 240 (otherwise the rest of the software will not work).
See also the FAQ for the 16 bit mode...

mulorc, 2019-07-17:
Thank you very much. Now is perfect. You are a good man.

olikraus, 2019-07-17:
:)

closed 2019-08-13 by olikraus

MartiX79, 2021-04-28:
Hi,
I have the same problem. I have enabled 16 Bit mode, but still the same issue.
My LCD is sdd1322 256x64 and as above, with U8x8 is ok but with U8G2 is not working - missing end pixels.
Can you help me, please?