olikraus/u8g2 #952: [Font size request] Support more size of vietnamese unicode font

lequangdau1993, 2019-07-18:
Hi Olikraus,

I'm using your library u8g2 for display content in Vietnamese.
Currently, My project able to display Vietnamese by using your lib successfully, but it only have one size (10 Pixel Height) supported.
Could you build a new one for supported smaller size on this font ?
It's would be nice if it supported from 6 to 8 pixel height. 

Sincerely,
Dau Le

https://github.com/olikraus/u8g2/wiki/fntgrpunifont

olikraus, 2019-07-18:
> Could you build a new one for supported smaller size on this font ?

I can only convert what you provide. This means: You need to provide a font in 6 to 8 pixel size, then I can convert this to the u8g2 internal format.

There might be a misunderstanding: unifont is a bitmap font (NOT a truetype font). It can NOT be scaled down to a smaller size. 
In fact truetype fonts can not be scaled down to 6 or 8 pixel size. In most cases they will look very ugly.
My suggestion is this: Please look in the internet for a suitable bitmap font. Post the link here, then I can convert the font for you.

Finding a bitmap font which supports Vietnamese might be indeed difficult. A quick research just revealed this font:
https://www.behance.net/gallery/49541299/SH-Pinscher-Free-Bitmap-Font
But this seems to be even bigger.

lequangdau1993, 2019-07-20:
Thanks for your info, I greatly appreciate your suggestion. 
From now, I'm going to dig more about it.

Thanks,
Dau Le

olikraus, 2019-08-13:
closing this for now. Please reopen, if you found a suitable font.

closed 2019-08-13 by olikraus