olikraus/u8g2 #953: C setup with SSD1309 pn STM32F446

DuyTrandeLion, 2019-07-19:
Hello,

I am using the 2.42" SSD1309 OLED + STM32F446 in SPI mode. 

Here is my OLED: https://www.aliexpress.com/item/32831554443.html?spm=2114.search0302.3.138.113b10d1njWwRJ&ws_ab_test=searchweb0_0%2Csearchweb201602_0_10084_10083_10547_10887_10307_321_453_10548_322_454_10618_536_10065_317_537_10068_319_10059_10103_10884_10696%2Csearchweb201603_0%2CppcSwitch_0&algo_pvid=35c7294a-4488-4c10-986b-ca050f24a086&algo_expid=35c7294a-4488-4c10-986b-ca050f24a086-16 

I use the following connection:
SDA -> PA7
SCL  -> PA5
RES  -> PA9
DC   -> PC7
CS   -> PB6

SPI clock: 22.5 MBits/s
CPOL = 0
CPHA = 0

There are nothing on the data bus and the clock bus. The display worked when I tested it with Arduino Uno. Could you give me a guidance to debug, or to make it work? Thanks.

> uint8_t u8x8_stm32_gpio_and_delay_callback(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)

{
    switch (msg)
    {
        case U8X8_MSG_GPIO_AND_DELAY_INIT:
        {
            HAL_Delay(1);
            break;
        }
        
        case U8X8_MSG_DELAY_MILLI:
        {
            HAL_Delay(arg_int);
            break;
        }

        case U8X8_MSG_GPIO_DC:
        {
            HAL_GPIO_WritePin(DC_GPIO_Port, DC_Pin, (GPIO_PinState)arg_int);
            break;
        }

        case U8X8_MSG_GPIO_RESET:
        {
            HAL_GPIO_WritePin(RES_GPIO_Port, RES_Pin, (GPIO_PinState)arg_int);
            break;
        }

        case U8X8_MSG_GPIO_CS:
        {
            HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, (GPIO_PinState)arg_int);
            break;
        }

        default: return 0;
    }
    return 1;  
}

uint8_t u8x8_stm32_4wire_spi_callback(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    switch (msg)
    {
        case U8X8_MSG_BYTE_SEND:
        {
            HAL_SPI_Transmit(&hspi1, (uint8_t *)arg_ptr, arg_int, 1000);
            break;
        }
        
        case U8X8_MSG_BYTE_INIT:
        {

            break;
        }

        case U8X8_MSG_BYTE_SET_DC:
        {
            HAL_GPIO_WritePin(DC_GPIO_Port, DC_Pin, (GPIO_PinState)arg_int);
            break;
        }

        case U8X8_MSG_BYTE_START_TRANSFER:
        {
            
            break;
        }

        case U8X8_MSG_BYTE_END_TRANSFER:
        {

            break;
        }

        default: return 0;
    }
    return 1;  
}

u8g2_Setup_ssd1309_128x64_noname2_2(&u8g2, U8G2_R0, u8x8_stm32_4wire_spi_callback, u8x8_stm32_gpio_and_delay_callback);
u8g2_InitDisplay(&u8g2); 
u8g2_SetPowerSave(&u8g2, 1);
u8g2_SetFont(&u8g2, u8x8_font_chroma48medium8_r);

olikraus, 2019-07-19:
For me it is difficult to support any none-Arduino solution. 
As a general statement I can say this:
The code above is incomplete and at least the CS related code is missing "U8X8_MSG_GPIO_CS" is not used and will never be executed.

DuyTrandeLion, 2019-07-19:
Hello,

I already have 
    case U8X8_MSG_GPIO_CS:
    {
        HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, (GPIO_PinState)arg_int);
        break;
    }
That's all I have from the example.

olikraus, 2019-07-19:
Sure, but the message U8X8_MSG_GPIO_CS has to be sent from somewhere. Usually this is the byte callback, but I do not see this in your example.

Thats what I am trying to say: You took something from someone who did something for U8g2. How can I support here? Maybe you should start from scratch and follow the guideline here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

olikraus, 2019-07-19:
I saw your post on elastic notes. For sure this is the right place to ask, but I guess the key point here is this sentence
"But if you will have some problems with communication, you can try to play with toggling this pin there."

So I assume the original author of elasic notes has no idea how to add CS here.

Maybe the following extension to your byte callback works:
```
case U8X8_MSG_BYTE_SEND:
{
u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
HAL_SPI_Transmit(&hspi1, (uint8_t *)arg_ptr, arg_int, 1000);
u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
break;
}
```

DuyTrandeLion, 2019-07-19:
Hello,
Thank you for your feedback and guidance. I will try and feedback on the problem within 24 hours.

closed 2019-07-22 by DuyTrandeLion

elektronek, 2020-03-26:
Hi olikraus,
thank you very much for solution, i'm running on STM32F103 and CS pin control not working,

in function gpio_and_delay have this line for control CS pin,

		case U8X8_MSG_GPIO_CS:
			HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, arg_int);
			break;

but without your recomend

		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level); 

and disable after send not working, CS pin is still in LOW state, now working correctly :-)

Best Regards!
Martin