olikraus/u8g2 #955: Undefined symbol u8g2_Setup_ssd1306_i2c_128x64_noname_1 (referred from main.o).

tpadcps999, 2019-07-25:
Hi , I using Nuvoton M252 MCU and try porting u8g2 library using keil C(V6.12)
I 'm follow "Porting to new MCU platform" but have some trouble 
when i build the project get follow error

.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_Setup_ssd1306_i2c_128x64_noname_1 (referred from main.o).

.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_cb_r0 (referred from main.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8x8_InitDisplay (referred from main.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8x8_SetPowerSave (referred from main.o).
but I already add all files of the u8g2/csrc to my project and inclue Paths

 this is my main.c

/****************************************************************************//**
 * @file     main.c
 * @version  V0.10
 * @brief
 *           Demonstrate how to set I2C Master mode and Slave mode.
 *           And show how a master access a slave on a chip.
 *
 * @copyright (C) 2018 Nuvoton Technology Corp. All rights reserved.
 *****************************************************************************/
#include <stdio.h>
#include "u8g2.h"
#include "NuMicro.h"

 u8g2_t u8g2;
/*---------------------------------------------------------------------------------------------------------*/
/* Global variables                                                                                        */
/*---------------------------------------------------------------------------------------------------------*/

volatile uint8_t g_u8MstEndFlag = 0;

typedef void (*I2C_FUNC)(uint32_t u32Status);

volatile static I2C_FUNC s_pfnI2C0Handler = NULL;
volatile static I2C_FUNC s_pfnI2C1Handler = NULL;
// Screenbuffer

/*---------------------------------------------------------------------------------------------------------*/
/*  u8g2                                                                                                   */
/*---------------------------------------------------------------------------------------------------------*/
uint8_t u8x8_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8,U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,U8X8_UNUSED void *arg_ptr)
{

			switch (msg)
			{

	
				case U8X8_MSG_GPIO_AND_DELAY_INIT:
				//HAL_Delay(1);
				TIMER_Delay(TIMER0, 1000); 
				break;
				case U8X8_MSG_DELAY_MILLI:
				//HAL_Delay(arg_int);
				TIMER_Delay(TIMER0, 1000*arg_int);
				break;
				case U8X8_MSG_DELAY_I2C:
				// delayUs(arg_int);
				break;
				case U8X8_MSG_GPIO_I2C_CLOCK:   
				// arg_int=0: Output low at I2C clock pin  
				// arg_int=1: Input dir with pullup high for I2C clock pin
				//HAL_GPIO_WritePin(GPIOB, GPIO_PIN_6, arg_int);
				if(arg_int==0)
				{	PB5=0;}
				else if(arg_int==1)
				{ PB5=1;}
				break;              
				case U8X8_MSG_GPIO_I2C_DATA:      
[... 195 lines cut ...]
    printf("| I2C Master (I2C0) <---> I2C Slave(I2C1)               |\n");
    printf("+-------------------------------------------------------+\n");

    printf("\n");
    printf("Configure I2C0 as Master, and I2C1 as a slave.\n");
    printf("The I/O connection I2C0 to I2C1:\n");
    printf("I2C0_SDA, I2C0_SCL\n");
    printf("I2C1_SDA, I2C1_SCL\n\n");

    /* Init I2C0 */
    I2C0_Init();
    
		u8g2_Setup_ssd1306_i2c_128x64_noname_1(&u8g2, U8G2_R0, u8x8_byte_hw_i2c, u8x8_gpio_and_delay);  

		u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
	  u8g2_SetPowerSave(&u8g2, 0); // wake up display
		int32_t k=0;
		while (1)
		{
			
		
		};

}
/*** (C) COPYRIGHT 2018 Nuvoton Technology Corp. ***/

olikraus, 2019-07-25:
The procedure is defined here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_d_setup.c#L244

tpadcps999, 2019-07-25:
Hi olikraus : Thank you 
I try to add "u8g2_d_stup.c" to my keil project but got more error like this ><
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_m_11_6_1 (referred from u8g2_d_setup.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_m_11_6_2 (referred from u8g2_d_setup.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_m_11_6_f (referred from u8g2_d_setup.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_m_12_12_1 (referred from u8g2_d_setup.o).
.\obj\I2C_Loopback.axf: Error: L6218E: Undefined symbol u8g2_m_12_12_2 (referred from u8g2_d_setup.o).

olikraus, 2019-07-25:
... which are all available here: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_d_memory.c

I am always happy to support here, but shouldn't it be clear to include all the files of the u8g2 project?

olikraus, 2019-07-25:
For possible future discussions in this thread
https://community.arm.com/developer/tools-software/tools/f/keil-forum/31840/static-library-links-differently-with-gcc-ld-and-armlink
http://www.keil.com/support/man/docs/armlink/armlink_pge1362075556280.htm

tpadcps999, 2019-07-25:
Olikraus, thank for immediate and proud support ,I'm include all .c first but ,flash was overflow
I will study linker problem first

olikraus, 2019-07-25:
Yes, I expected that. You need to activate the garbage collector during link step. See my links above.

tpadcps999, 2019-07-25:
Olikraus, thank for help ,  In keil "Options for Target" -> "c/c++" -> Click "One ELF Section per Function"
 that can remove unused .c file  .
And  my first project working now

[image]

olikraus, 2019-07-25:
:)

closed 2019-08-13 by olikraus