olikraus/u8g2 #956: u8g2_SetFont(..) fails with ...region `ram' overflowed by XXXX bytes

samsam4, 2019-07-25:
WICED6.2.1 with CYW20719 fails on u8g2_SetFont(..) with:

Linking target ELF
..\..\43xxx_Wi-Fi\tools\ARM_GNU\Win32\bin\arm-none-eabi-ld: ../build/bsc01-CYBT_413034_EVAL-rom-ram-Wiced-release/A_20719B1-bsc01-rom-ram-spar.elf section `.rodata' will not fit in region `ram'
..\..\43xxx_Wi-Fi\tools\ARM_GNU\Win32\bin\arm-none-eabi-ld: region `ram' overflowed by 126956 bytes

I found this [posting](https://github.com/olikraus/u8g2/issues/668#issuecomment-414266601) and this [example makefile](https://github.com/olikraus/u8g2/blob/master/sys/arm/lpc824/u8g2_logo/Makefile) 
Is there anything different from what is posted before for STM , I need to make in order to make WICED/CYW20719 makefile.mk to work with u8g2 or u8x8  ?
Thanks

olikraus, 2019-07-25:
The toolchain is the same. 
You need to use 
-ffunction-sections -fdata-sections (compile step)
and
-Wl,--gc-sections (linking step)

samsam4, 2019-07-25:
I already add in the projects makefile.mk 
C_FLAGS += -ffunction-sections -fdata-sections

still cant figure yet how to add **-Wl,--gc-sections**  (many hierarchical makefile-s) :( 
but the code works so far even without the last flags :)
Thanks

olikraus, 2019-07-25:
Ok...

closed 2019-08-13 by olikraus