olikraus/u8g2 #961: Pins for second I2C (Heltec WiFi LoRa 32 (V2))

SALLDHMF, 2019-07-27:
Hello again :-)

Continuation of my learning about Heltec WiFi LoRa 32 (V2) #942. Now I am at the stage of connecting the second display to this module. The first display, embedded in the Heltec WiFi LoRa 32 (V2) module, works great - U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2 (U8G2_R0, 16, 15, 4).

The second (U8G2_SSD1327_EA_W128128, U8G2_SSD1327_MIDAS_128X128 or U8G2_SSD1327_WS_128X128) display works fine through SW, but of course very slowly and I need to switch it to HW. The I2C scanner from Heltec shows that the address of the display on the I2C1 ESP32 bus (Wire1) is 0x3C.

ESP32 has the ability to use hardware I2C on any GPIO. However, the use of ..._ 2ND_HW_I2C ... does not give the possibility to indicate specific ports of the module to which the display is connected. But... descriptions of similar problems (#483, #779) are for me ... "too clever" ;-). Maybe there is a simpler method than editing the h and/or cpp file code?

---
https://heltec.org/project/wifi-lora-32/
https://github.com/HelTecAutomation/Heltec_ESP32
https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/LoRa/LoRaReceiver/LoRaReceiver.ino

Tomasz

dilucide, 2019-07-28:
Are you by any chance seeing a message "no Default SDA Pin for Second Peripheral" on the ESP32s UART?

The reason I am asking is that, looking at the ESP32 Arduino Wire implementation, it seems as though there is no default Pin defined for the second HW I2C. In u8x8_byte_arduino_2nd_hw_i2c, Wire1.begin(...) is called without arguments, thus an error is expected.

Currently, there seems to be no solution without editing some of the library files. In my opinion, the solution in #779 seems to be the most elegant.

SALLDHMF, 2019-07-28:
Thanks for the answer. I did not see the message "no Default SDA Pin for Second Peripheral".

I probably should not start something that I can not do. Finding files and changing fragments according to # 779 is feasible even without knowing what I'm changing. I do not know how my sketch should look after this change. Certainly, the constructor must change, but how?

SALLDHMF, 2019-07-29:
OK. Let's change something in these files :-).

In SW version, the display works as SSD1327_MIDAS_128X128 - U8G2_SSD1327_MIDAS_128X128_F_SW_I2C u8g2 (U8G2_R0, 17, 2). I made the following changes
in u8g2lib.h it was:
```
class U8G2_SSD1327_MIDAS_128X128_F_2ND_HW_I2C: public U8G2 {
   public: U8G2_SSD1327_MIDAS_128X128_F_2ND_HW_I2C (const u8g2_cb_t * rotation, uint8_t reset = U8X8_PIN_NONE): U8G2 () {
     u8g2_Setup_ssd1327_i2c_midas_128x128_f (& u8g2, rotation, u8x8_byte_arduino_2nd_hw_i2c, u8x8_gpio_and_delay_arduino);
     u8x8_SetPin_HW_I2C (getU8x8 (), reset);
   }
};
```
and now it is:
```
class U8G2_SSD1327_MIDAS_128X128_F_2ND_HW_I2C: public U8G2 {
   public: U8G2_SSD1327_MIDAS_128X128_F_2ND_HW_I2C (const u8g2_cb_t * rotation, uint8_t reset = U8X8_PIN_NONE, uint8_t clock = U8X8_PIN_NONE, uint8_t data = U8X8_PIN_NONE): U8G2 () {
     u8g2_Setup_ssd1327_i2c_midas_128x128_f (& u8g2, rotation, u8x8_byte_arduino_2nd_hw_i2c, u8x8_gpio_and_delay_arduino);
     u8x8_SetPin_HW_I2C (getU8x8 (), reset, clock, data);
   }
};
```
I changed the U8x8lib.cpp file as in # 779. The compilation does not show errors, but I do not have anything on the display.

olikraus, 2019-07-29:
One precondition for the second I2C interface is a proper definition of `WIRE_INTERFACES_COUNT` by the ESP32. The value for `WIRE_INTERFACES_COUNT` must be 2 or higher, denoting, that there is an additional Wire object called "wire1".

https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L113

If this interface count is only 1, then the 2nd wire interface callback evaluates to the empty function (and obviously will not do anyting): https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L1182

EPS32 defines a Wire1 object: https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L137
However, it does not tell, that such an object exists.

Compare, for example STM32 Arduino software: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/WireSlave/src/Wire_slave.h#L37

As a workaround we could check for ESP32 environment (question: How to do this?) and enable the 2nd I2C interface.

Maybe this works. After this line
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L130
add the following code:
```
#if defined(ARDUINO_ARCH_ESP32)
#define U8X8_HAVE_2ND_HW_SPI
#endif     
```

SALLDHMF, 2019-07-29:
Thanks for the answer. I am talking simultaneously with hunter8801 from Heltec - producer of WiFi LoRa 32 module (V2). Maybe when we combine your knowledge, it will enable us to solve the problem of the second display.

http://community.heltec.cn/t/the-hardware-i2c-for-second-display-with-wifi-lora-32-v2/150/10

olikraus, 2019-08-13:
I want though th discussion above. I think the discussion went a little bit into the wrong direction.
It is no problem to use u8g2 to drive two different displays with two different wire object (BTW: you could also driver two displays via the same wire object using the same bus but different I2C addresses).

Anyhow, the key problem is: How can U8g2 know that there is a second wire object called (wire1)?

This can not be answered by haltec, instead this is a problem of the ESP32 Arduino port.

You probably need to create an issue for https://github.com/espressif/arduino-esp32 and ask for WIRE_INTERFACES_COUNT

closed 2019-10-03 by olikraus

dstulken, 2021-02-26:
Any chance of revisiting this issue?

ESP32 Arduino still doesn't have a WIRE_INTERFACES_COUNT definition, so for us U8g2 users, 2nd I2C bus connections are still non-functional, even after patching the library as in issue #779.

I think it would be a one-line change - in U8x8lib.cpp, in the u8x8_byte_arduino_2nd_hw_i2c() function, the line:
`#ifdef U8X8_HAVE_2ND_HW_I2C` 
changes to 
`#if defined(U8X8_HAVE_2ND_HW_I2C) || defined(ARDUINO_ARCH_ESP32)`
... and that does the trick.

I certainly understand the desire to have upstream sources standardize/unify in the way they present information...  but if it hasn't happened in a year and a half, it probably isn't going to.  I think the answer to "How can U8g2 know?" is that we can know because it's an ESP32.  It's platform-specific code, yes, but so is the Wire1.begin() code just a few lines below that...

Thanks