olikraus/u8g2 #962: Formating text to multiple lines

djairjr, 2019-07-30:
Hello. I developed this routine to split a long String and display it on multiple lines of the ST7920 display using Arduino Mega.
Hope it can be useful.

Ps.: I'm still not using the word count routine (through the number of spaces in sentence + 1), but the idea is to use it for hyphenation or better word distribution on the display.

`
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define ST7920_CK     16      // PINOUT for DISPLAY
#define ST7920_DATA   17      // 128x64 - ST7920
#define ST7920_CS     18      // USING ARDUINO MEGA
#define ST7920_RST    19      

U8G2_ST7920_128X64_F_SW_SPI pager (U8G2_R0, ST7920_CK, ST7920_DATA, ST7920_CS, ST7920_RST);

void setup() {
  Serial.begin (115200); // Just for debugging
  pager.begin();
  pager.setBitmapMode (1);
  pager.enableUTF8Print();
  pager.setFlipMode(0);
  pager.setFontMode(0);
  pager.setDrawColor(1);
  pager.setFontDirection(0);
  pager.setFont(u8g2_font_5x7_tf );
}

void loop() {
  pager.firstPage();
    do {
    formatMessage ("THIS COULD BE ANY LONG SENTENCE AND MY PROGRAM WILL WORK?", 10, 10, 128, 64);
    delay(100);
  } while ( pager.nextPage() );
}

void formatMessage(String message, int col, int line, int maxcol, int maxline) {

  int window_width = maxcol - col;
  int window_height = maxline - line;

  int font_width = pager.getMaxCharWidth();
  int font_height = pager.getMaxCharHeight() + 1;

  int amountline = round (window_height / font_height);
  int amountchar = round (window_width / font_width);

  String newmessage = "";
  int sizeMessage = message.length() - 1;

  int howMuchLine = round (sizeMessage / amountchar);
  int index;
  int lastLine = sizeMessage % amountchar; // Need another line?
  int linePosition;

  if (lastLine > 0) {
    howMuchLine = howMuchLine + 1;
  }

  if (sizeMessage > amountchar) {
    for (int l = 1; l <= howMuchLine; l++) {
      linePosition = l;
      for (int c = 1; c <= amountchar; c++) {
        index = amountchar * (l - 1) + (c - 1);
        if (index <= sizeMessage) {
          newmessage = newmessage + message[index];
        }
      }
      int lenNew = newmessage.length() + 1;
      char splitted[lenNew];
      newmessage.toCharArray(splitted, lenNew);
      pager.drawUTF8 (col, (((font_height) * linePosition) + line) , splitted);
      newmessage = "";
    }
  } else { 
    char splitted[message.length()+1];
    message.toCharArray(splitted, message.length()+1);
    pager.drawUTF8 (col, line , splitted);
  }
}

int wordcount (String message) {
  int startIndex = 0;
  int countword = 0;
  while (message.indexOf (" ", startIndex) != -1) {
    startIndex = message.indexOf (" ", startIndex) + 1;
    countword += 1;
  }
  countword += 1;
  return (countword);
}
`

olikraus, 2019-07-30:
Sounds good. Maybe code tags would be nice.

closed 2019-08-13 by olikraus

leifclaesson, 2025-04-26:
Not very useful, it uses the widest character as the basis for the calculation so it only works with monospaced fonts.

olikraus, 2025-05-01:
Very old topic... anyhow I fixed the code tags...