olikraus/u8g2 #963: setFontDirection does not print in expected direction for (2)

prabbit237, 2019-07-31:
Using setFontDirection(0) orients text normally or as expected. From the image on the wiki, using (1) turns it sideways and prints top-to-bottom, which makes sense (it's basically like rotating that section by 90 degrees.) But setFontDirection(2) is NOT as you would expect. Instead of printing "Hello, world" upside down (so if you physically rotated the screen 180 degrees, it'd show normal) it prints **"dlrow ,olleH"** (and prints it upside down, as expected, unless/until the screen is rotated.) 

I have a script that uses an accelerometer (and strange that Firefox spell check didn't have that word in it already) that reverses some stuff when the screen is physically rotated. So when it's flipped, I have to tell it to:

      if (flipped){
         display.setFontDirection(2);
         display.print("dlrow ,olleH");
      }else{
         display.setFontDirection(0);
         display.print("Hello, world");
      }

What I'd EXPECT to be the case is that I'd tell it:

      if (flipped){
         display.setFontDirection(2);
      }else{
         display.setFontDirection(0);
      }
      display.print("Hello, world");

and it would then look the same when it's flipped or not flipped and not as "dlrow ,olleH" in one case and as "Hello, world" in the other. So for what I'm wanting to print, I have to use separate .setCursor() and .print() commands for each character (since it's not static strings) to get them into the correct order.

[image]

prabbit237, 2019-07-31:
BTW, I realize changing the library may break other user's code so it's more of a "can we get this documented in the wiki?" than a "can you fix this?"

olikraus, 2019-07-31:
Hmmm... Can you provide a full working example? What is your constructor?

prabbit237, 2019-07-31:
I had a feeling you was gonna ask for that          **laugh**

Ok, here's code samples and photos. This is going to be for a coffee timer at work so it basically cycles between an hourglass (modified from https://github.com/szczys/sand-ular-automata) and the smiliys (based on the time/coffee quality.) It'll be in a box that can be flipped to restart the timer, thus the accelerometer. The code for the hourglass doesn't care about orientation (but there is one   display.setDisplayRotation(U8G2_R3) line in setup because it's written for a 64x128 display and I'm using a 128x64.) So the subsequent flips are all due to my text/images being added.

Normal orientation code (with all the extra stuff pulled out except for the smilies, which are in a separate header not included.)

    #include <Wire.h>
    #include <U8g2lib.h>
    #include "face1.h"

    U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

    // the setup routine runs once when you press reset:
    void setup() {
      display.begin();
      display.setDisplayRotation(U8G2_R3);
      display.setFontRefHeightExtendedText();
      display.setDrawColor(1);
      display.setFontPosTop();
      display.setFontDirection(0);
      display.setFont(u8g2_font_7x13_tf);
    }

    // the loop routine runs over and over again forever:
    void loop() {
           display.clearBuffer();
               display.drawXBM(0, 0, face1_width, face1_height, face1_bits);
               display.setFontDirection(0);
               display.setFontMode(1);
               display.setCursor(17, 74);
               display.print("*UGH*");
               display.setFont(u8g2_font_6x10_tr);
               display.setCursor(0, 88);
               display.print("Coffee");
               display.setCursor(39, 88);
               display.print("good");
               display.setCursor(0, 103);
               display.print("Me");
               display.setCursor(14, 103);
               display.print("want");
               display.setCursor(40, 103);
               display.print("some");
               display.setCursor(1, 118);
               display.print("Wake me up");
           display.sendBuffer();
           while (1)
             delay(1);
    }

Flipped code:

    #include <Wire.h>
    #include <U8g2lib.h>
    #include "iface1.h"

    U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

    // the setup routine runs once when you press reset:
    void setup() {

     display.begin();
     display.setDisplayRotation(U8G2_R3);
     display.setFontRefHeightExtendedText();
     display.setDrawColor(1);
     display.setFontPosTop();
     display.setFontDirection(0);
     display.setFont(u8g2_font_7x13_tf);

    }

    // the loop routine runs over and over again forever:
    void loop() {
           display.clearBuffer();
           display.clearBuffer();
           display.drawXBM(0, 128 - iface1_height, iface1_width, iface1_height, iface1_bits);
           display.setFontDirection(2);
           display.setFontMode(1);
               display.setCursor(18, 54);
               display.print("*HGU*");
               display.setFont(u8g2_font_6x10_tr);
               display.setCursor(5, 40);
               //        display.print("Coffee good");
               display.print("doog");
               display.setCursor(32, 40);
               display.print("eeffoC");
               display.setCursor(4, 25);
               //        display.print("Me want some");
               display.print("emos");
               display.setCursor(30, 25);
               display.print("tnaw");
               display.setCursor(56, 25);
               display.print("eM");
               display.setCursor(5, 10);
               //        display.print("Wake me up");
               display.print("pu em ekaW");
           display.sendBuffer();
           while (1)
             delay(1);
    }

If I used "display.print("*UGH*");" in the inverted one, it comes out as "*HGU*" (letters are not mirrored and look fine if I flipped the display around 180 degrees. It's just the order of them is backwards.)

Note in the images, I left the board in the same position. You can see the normal one uses the correct spelling but in the other, I had to reverse all the strings. I did also invert the smilies when saving them and use two different bitmaps with different offsets (the "iface1.h" and "face1.h", etc differences aren't that obvious to spot.)

[image]
[image]

prabbit237, 2019-07-31:
The more I look at it, the more I believe is happening is:

    setDisplayRotation(x);  //rotates the whole screen around
    setCursor(x,y);             //anchors the corner of the tile for the letter.
    setFontDirection(x);     //pivots the letter around the anchor point of the upper-left corner
                                       //So far, so good.
    print(string);                //This now does something like:
                                       //for (i=0; i<len(string);i++)   drawGlyph(string[i]);
                                       //So if it's anchored to/offset from the **left** side with the setCursor()
                                       //then it's printing the chars in the correct left-right order
                                       //but I'm expecting it to be for (i=len(string);i>0;i--)   display(string[i-i]);
                                       //where the string starts from from the **right** side even though the 
                                       //anchor is on the left.

So you can chalk it up to maybe me misusing the code somewhat but it's still not intuitive in that operation and could at least use some explanation in the wiki.

olikraus, 2019-08-02:
I have to postpone this topic a little bit due to some personal constrains...

Meanwhile if you can create a small minimal but complete example which shows the mirrored text, that would be great.

prabbit237, 2019-08-03:
OK, here's a bare-bones script showing what happens:

    #include <U8g2lib.h>
    U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

    void setup() {
      display.begin();
      display.setFont(u8g2_font_7x13_tf);
      display.clearBuffer();
      display.setFontDirection(2);
      display.setCursor(15, 20);
      display.print("Test");
      display.setFontDirection(0);
      display.setCursor(10, 60);
      display.print("Test");
      display.sendBuffer();
    }
    void loop() {
    }

Result of that is:

[image]

And here's what I was talking about with expecting it'd anchor in one corner and rotate around that (editing the sample from the wiki and showing what I'd expected (2) and (3) to do based on (0) and (1) positions.) 

[image]

But I did realize that with the code I was working on, since my text and the hourglass are in separate frames, I could just rotate the screen one way, display the text/smiley and then rotate it back again and switch to the hourglass (it shows the hourglass for about three seconds and the other for one, switching back and forth) so I'm happy with that work-around for this project at least. And the library does work out nicely for everything else I've used it with so I do applaud the hard work on it.

olikraus, 2019-08-04:
Thanks for the minimal example.

olikraus, 2019-08-04:
Did you try drawStr instead of 'print'? There might be a bug in the print fn.

prabbit237, 2019-08-04:
Good question and when I first saw it, I asked myself "WHY didn't I try drawStr?" (I simply knew I hadn't.) And sure enough drawStr does work as expected for strings but................

Where I first saw the issue was when I was trying to display a number and under drawStr in the wiki, it says "Note 2: Use the print function to print the value of a numeric variable." So yes, for something that is a char[] variable, drawStr would be the simplest/best way to do it but for numbers it's a no-go.  I understand it may not be an easily-fixed bug but at least the wiki should be updated :) 

There's definitely ways of converting a number to a string and then drawStr would work fine (but in my case, I was only printing a time-remaining in the format of h:mm with a max of 2 hours so I just parsed the digits out and printed them separately so it wasn't a huge deal.)

prabbit237, 2019-08-04:
Well, to quote the "as seen on TV" commercials, "but wait! There's MORE!" (or "it goes from bad to worse.")

I tested the other settings for the direction with the following code:

    #include <U8g2lib.h>
    U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

    void setup() {
      display.begin();
      display.setFont(u8g2_font_7x13_tf);
      display.clearBuffer();
      display.setFontDirection(0);
      display.setCursor(10, 10);
      display.print("Test 0");
      display.setFontDirection(1);
      display.setCursor(10, 20);
      display.print("Test 1");
      display.setFontDirection(2);
      display.setCursor(10, 30);
      display.print("Test 2");
      display.setFontDirection(3);
      display.setCursor(10,50);
      display.print("Test 3");
      display.sendBuffer();}

    void loop() {
    }

Results:

[image]

It's a bit hard to see but (1) and (3) didn't even space the chars out correctly. Look at the "Te" on (1) and the "st" on (3) and you can see they overlap some. 

So basically the wiki should just have a note reading something like "this does not work with print(). It only works with drawStr()." (And I just now realized the example in the wiki only used drawStr so that it does wind up going back to my initial remark about how maybe I was just doing things wrong or that the wiki wasn't quite fully documenting it clearly.)

olikraus, 2019-08-04:
I have only limited access to my repo. I can not do further tests at the moment. I need to see to which extend I can fix the print function. Indeed print is the original function from Arduino serial.print.

olikraus, 2020-01-19:
Fixing this will add 50 bytes (on AVR) to the lib.
If `U8G2_WITH_FONT_ROTATION` is commetned, then u8g2 size will be reduced by 600 bytes (avr).

olikraus, 2020-01-19:
done

closed 2020-01-19 by olikraus