olikraus/u8g2 #965: OLED display i2c on STM32

sfrwmaker, 2019-08-03:
Dear sir!

I realy appreciate the product you have created and am using it in my projects for a long time. Now am going to implement your u8g2 library on new project based on STM32 micro controller and i2c OLED display. I have read 'Porting to new MCU platform' guide and still trying to create my own callbacks functions. 
SPI version of the display works fine on STM32, but i have issues with i2c version of display.

Please, look the files attached. The main problem is in u8x8_byte_stm32_hw_i2c() callback.
I am collecting the data into local buffer on 'U8X8_MSG_BYTE_SEND' event and am transfing the data to the OLED display on 'U8X8_MSG_BYTE_START_TRANSFER' event.
On first 'U8X8_MSG_BYTE_START_TRANSFER' event i have 25 bytes long array of data:
174,213,128,168,63,211,0,64,141,20, 32,0,161,200,218,18,129,207,217,241, 219,64,46,164,166.
On second event i have 4-bytes: 64,16,2,176 and i have the same 4-bytes arrays next times. It looks like a screen data.

If i use Arduino platform, i have the same codes like on first event but they are prefixed by zero bytes (command byte for OLED).
I have investigated OLED datasheet and found that message to the OLED display can be in two forms:
1. <address><command><data>
2. <address><command><data><data>...
Suppose, first sequence should start with leading zero byte.
Second enevt starts with byte 64 (0x40 - DATA BLOCK) but it followed by only 3 bytes.

On arduino platform i have met byte 64 followed by 24 other bytes.
It looks like callback procedure does not reseive the command bytes. 
Can you help?

[sources.zip](https://github.com/olikraus/u8g2/files/3463907/sources.zip)

sfrwmaker, 2019-08-03:
Hello!

I have fixed the problem. I have thought than SPI and i2c callback should work the same way, distinguishing between command and data. And DC event can be used to check it.
Everything is working like a charm. 

uint8_t u8x8_byte_stm32_hw_i2c(u8x8_t *u8g2, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	static uint8_t dc = 0;

	switch(msg)  {
		case U8X8_MSG_BYTE_SEND:
			while (HAL_I2C_STATE_READY != HAL_I2C_GetState(&I2C_HANDLER)) { }
			HAL_I2C_Mem_Write(&I2C_HANDLER, OLED_I2C_ADDR<<1, (dc == 0)?0:0x40, 1, (uint8_t *)arg_ptr, arg_int, HAL_MAX_DELAY);
			break;
		case U8X8_MSG_BYTE_INIT:
			u8g2->gpio_and_delay_cb(u8g2, U8X8_MSG_DELAY_NANO, u8g2->display_info->post_chip_enable_wait_ns, NULL);
			break;
		case U8X8_MSG_BYTE_SET_DC:
			dc = arg_int;
			break;
		case U8X8_MSG_BYTE_START_TRANSFER:
			break;
		case U8X8_MSG_BYTE_END_TRANSFER:
			break;
		default:
			return 0;
	}
	return 1;
}

closed 2019-08-03 by sfrwmaker

olikraus, 2019-08-03:
Hmmmm I do not know much about stm32 Hal...

electrotehnik, 2020-04-23:
I also connect SSD1306 to STM32103 in the HAL + FreeRTOS project. The library of the previous generation u8g worked perfectly in this system.

I need hardware I2C. And trying to write a constructor:
- u8x8_byte_stm32_hw_i2c
- u8g2_gpio_and_delay_stm32

> uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	  switch (msg)
	  {
	  case U8X8_MSG_GPIO_AND_DELAY_INIT:
	    HAL_Delay(1);
	    break;
	  case U8X8_MSG_DELAY_MILLI:
	    HAL_Delay(arg_int);
	    break;
	  case U8X8_MSG_GPIO_DC:
	    //HAL_GPIO_WritePin(OLED_DC_GPIO_Port, OLED_DC_Pin, arg_int);
	    break;
	  case U8X8_MSG_GPIO_RESET:
	    //HAL_GPIO_WritePin(OLED_RES_GPIO_Port, OLED_RES_Pin, arg_int);
	    break;
	  }
	  return 1;
}

> uint8_t u8x8_byte_stm32_hw_i2c(u8x8_t *u8g2, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	static uint8_t dc = 0;
	switch(msg)  {
		case U8X8_MSG_BYTE_SEND:
			while (HAL_I2C_STATE_READY != HAL_I2C_GetState(&I2C_HANDLER)) { }
			HAL_I2C_Mem_Write(&I2C_HANDLER, DEVICE_ADDRESS, (dc == 0)?0:0x40, 1, (uint8_t *)arg_ptr, arg_int, HAL_MAX_DELAY);
			break;
		case U8X8_MSG_BYTE_INIT:
			u8g2->gpio_and_delay_cb(u8g2, U8X8_MSG_DELAY_NANO, u8g2->display_info->post_chip_enable_wait_ns, NULL);
			break;
		case U8X8_MSG_BYTE_SET_DC:
			dc = arg_int;
			break;
		case U8X8_MSG_BYTE_START_TRANSFER:
			break;
		case U8X8_MSG_BYTE_END_TRANSFER:
			break;
		default:
			return 0;
	}
	return 1;
}

The display is initiated, but shows only noise.

Could you help me write the constructors correctly?

olikraus, 2020-04-23:
I did not wrote the HAL procedure, so I can not say much about your code, except that it looks a little bit strange.

I once wrote code HAL code for an LPC804 (Cortex-M0):

https://github.com/olikraus/lpc804_plu/blob/8a99d231dfc91b4e7d65baf75b0f6a8e7167f63c/lpc804/u8g2_logo/u8x8cb.c#L138-L176

My own code follows the second example described here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

electrotehnik, 2020-04-23:
Thanks for the quick response!
I carefully study the examples you have proposed.

Here is the code with HAL that worked in the previous version of the library:

> uint8_t u8g_com_hw_i2c_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr)
 {
  switch(msg)
  {
  case U8G_COM_MSG_STOP:
    break;
   case U8G_COM_MSG_INIT:
    u8g_MicroDelay();
    break;
   case U8G_COM_MSG_ADDRESS:           /* define cmd (arg_val = 0) or data mode (arg_val = 1) */
    u8g_10MicroDelay();
    if (arg_val == 0)
    {
         control = 0;
    }
    else
    {
         control = 0x40;
    }
    break;
   case U8G_COM_MSG_WRITE_BYTE:
   {
        uint8_t buffer[2];
        buffer[0] = control;
        buffer[1] = arg_val;
        HAL_I2C_Master_Transmit(&I2C_HANDLER, DEVICE_ADDRESS, (uint8_t*) buffer, 2, I2C_TIMEOUT);
   }
        break;
   case U8G_COM_MSG_WRITE_SEQ:
   case U8G_COM_MSG_WRITE_SEQ_P:
   {
        uint8_t buffer[DATA_BUFFER_SIZE];
           uint8_t *ptr = arg_ptr;
           buffer[0] = control;
           for (int i = 1; i <= arg_val; i++)
           {
                buffer[i] = *(ptr++);
           }
           HAL_I2C_Master_Transmit(&I2C_HANDLER, DEVICE_ADDRESS, (uint8_t *)buffer, arg_val, I2C_TIMEOUT);
   }
    break;
  }
  return 1;
 }

olikraus, 2020-04-24:
The HAL function for the u8glib will not work with u8g2. You have to rewrite the HAL functions. Use the referenced template for this.

electrotehnik, 2020-04-24:
Thank you very much, olikraus!
I was able to run all with ST HAL driver on STM32F103CB, valid code below:

```
uint8_t u8x8_byte_stm32_hw_i2c(u8x8_t *u8g2, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	/* add extra byte for the i2c address */
	static uint8_t buffer[34];
	static uint8_t buf_idx;
	uint8_t *data;
	switch(msg)  {
		case U8X8_MSG_BYTE_SEND:
		    data = (uint8_t *)arg_ptr;
		    while( arg_int > 0 )
		    {
		    	buffer[buf_idx++] = *data;
		    	data++;
		    	arg_int--;
		    }
			break;
		case U8X8_MSG_BYTE_INIT:
			/* add your custom code to init i2c subsystem */
			break;
		case U8X8_MSG_BYTE_SET_DC:
			/* ignored for i2c */
			break;
		case U8X8_MSG_BYTE_START_TRANSFER:
			buf_idx = 0;
			break;
		case U8X8_MSG_BYTE_END_TRANSFER:
			HAL_I2C_Master_Transmit(&I2C_HANDLER, DEVICE_ADDRESS, (uint8_t *)buffer, buf_idx, I2C_TIMEOUT);
			break;
		default:
			return 0;
	}
	return 1;
}
```